platformio/platformio-core #3542: ValueError: path is on mount 'C:', start on mount 'd:

olonsoft, 2020-06-03:
When I click the Clean button (PlatformIO: Clean) I get the following error:
```
ValueError: path is on mount 'C:', start on mount 'd:':
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 133: 
    env.PioClean(env.subst("$BUILD_DIR"))
  File "C:\Users\irakl\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 299:
    clean_rel_path = relpath(clean_dir)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\ntpath.py", line 562:
    path_drive, start_drive))
```
My project compiles and run normally. Only "Clean" command reports this error.

olonsoft, 2020-06-03:
Yes
`build_dir=C:\programming\pio\build`

valeros, 2020-06-16:
Hi @olonsoft ! Thanks for the report. Please upgrade to the latest PlatformIO from the development branch via `pio upgrade --dev` and try again.

commit 21f3dd11 (2020-06-16): Fix printing relative paths on Windows // Resolve #3542

Fixes "ValueError" when running "clean" target if "build_dir"
points to a folder on a different logical drive
closed 2020-06-16 by valeros