platformio/platformio-core #3544: `--target` with no arguments does not produce list of targets

weberw, 2020-06-04:
### Configuration

**Operating system**: all

**PlatformIO Version** (`platformio --version`): 4.3.4

### Description of problem
According to https://docs.platformio.org/en/stable/projectconf/section_env_build.html#targets running `platformio run --target` should produce a list of targets, however an error is generated instead

#### Steps to Reproduce

1. `platformio run --target`

### Actual Results
```
Error: --target option requires an argument
```
### Expected Results
A list of build targets

### If problems with PlatformI

jnr2820, 2020-06-04:
I'm seeing this too.

```
pio run --target
Error: --target option requires an argument

pio --version
PlatformIO, version 4.3.4
```

ivankravets, 2020-06-09:
Thanks! Indeed, this feature has not been implemented yet. I renamed this issue.

closed 2020-06-22 by ivankravets

ivankravets, 2020-06-22:
Please re-test with `pio upgrade --dev` version of PIO core. See https://docs.platformio.org/en/latest/core/userguide/cmd_run.html#cmdoption-platformio-run-list-targets

jnr2820, 2020-06-26:
It's working for me. (Also, it is very neat. Thank you)
`
┌─[jnr2820@nano-cube-black] - [~/blink_machine] - [Fri Jun 26, 03:49]
└─[$]> platformio run --list-targets
Environment       Group     Name        Title                    Description
----------------  --------  ----------  -----------------------  -----------------------------------------------------
nanoatmega328new  Advanced  compiledb   Compilation Database     Generate compilation database 'compile_commands.json'
nanoatmega328new  Generic   clean       Clean
nanoatmega328new  Platform  bootloader  Burn Bootloader
nanoatmega328new  Platform  fuses       Set Fuses
nanoatmega328new  Platform  program     Upload using Programmer
nanoatmega328new  Platform  size        Program Size             Calculate program size
nanoatmega328new  Platform  upload      Upload
nanoatmega328new  Platform  uploadeep   Upload EEPROM
`

commit 9f05519c (2020-06-22): List available project targets with a new "platformio run –list-targets" command // Resolve #3544