platformio/platformio-core #3545: Warning! You are not connected to the Internet.

jumblies, 2020-06-04:
What kind of issue is this?

- [X] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [X] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
    

- [X] **PlatformIO Core**.
      4.3.4
------------------------------------------------------------------

### Configuration

**Docker**:

** Latest ** (`platformio --version`):
 4.3.4
### Description of problem
YOU ARE NOT CONNECTED TO THE INTERNET

#### Steps to Reproduce
1. Tried to compile in docker container on host machine
2. Tried to compile in docker container on remote machine
3. Tried to compile natively (without docker) on host machine
4. Tried to ping platformio.org with no response
```$ ping platformio.org
PING platformio.org (88.198.170.159): 56 data bytes```
5. Built container on Play with Docker with same error
6. Tried to ping from play with docker with same timeout

### Actual Results
```HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.```

### Expected Results
Successful compilation

manuelbarbosa, 2020-06-04:
Having the exact same issue, compiling ESPHome through HomeassistantOS (fresh install) with the 1.14.3 version. Can't ping platformio.org or api.platformio.org.

jumblies, 2020-06-04:
I suspect it is almost certainly a server or DNS issue on their side.  Naturally, it only happens when I need to compile urgently for a node failure.

tigattack, 2020-06-04:
In Home Assistant ESPHome, PlatformIO is pinned to version 4.0.3, which is using hardcoded IP addresses in [util.py](https://github.com/platformio/platformio-core/blob/v4.0.3/platformio/util.py#L381-L384).   
This is in fact _still_ the case in the latest develop branch of PlatformIO - [see here](https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L365-L368).   
I discovered this information [here](https://github.com/esphome/issues/issues/1265).

Can anyone answer why on earth hardcoded IPs are being used instead of DNS names?

jumblies, 2020-06-04:
Sounds like a dev branch that went into production or maybe they have bought a static IP.  

I'm going to close this issue since esphome/issues#1265 has more info

closed 2020-06-04 by jumblies

tigattack, 2020-06-04:
This has been part of the develop branch for a long time.   
Owning a static IP doesn't mean you can't use DNS names, which would solve exactly this sort of issue. Seems like a very poor decision.

anthonylavado, 2020-06-04:
> Can anyone answer why on earth hardcoded IPs are being used instead of DNS names?

While I didn't write it, the code only has to reach at least one of those in order to "pass". Since they've added `github.com` and `platformio.org` at the bottom of the list, it _should_ eventually succeed.