platformio/platformio-core #3548: ANSI colour codes not being properly processed in serial monitor

pfeerick, 2020-06-06:
- [*] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------
### Configuration
**Operating system:** Windows 10
**PlatformIO Version:** 4.3.4
**Python Version:** 3.7.7

### Description of problem
When using ANSI escape codes, ANSI output is not reliably displayed in there serial monitor, when either the `colorize` or `direct` `monitor_filters` options are used, or the `monitor_flags = --raw`. I've not repeated this on Linux (yet) to see if it's a more universal issue or Windows specific. It doesn't appear to matter whether `\e` or `\033` is used. However, adding a leading space before ` \e` if it is the first 'character' of a line seems to give more reliable results. Perhaps a character is being dropped somewhere?

#### Steps to Reproduce
1. Use the following `platformio.ini`:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_speed = 460800
monitor_filters = direct
```
2. Use the following code:
```
#include <Arduino.h>

void setup() {
  Serial.begin(9600);
  Serial.println();
  Serial.println("\033[33msketch init!\033[0m");
}

void loop() {
  Serial.println("\033[32mgreen\033[0m");
  Serial.println("\033[34mblue\033[0m");
  delay(1000);
}
```
3. Build, upload, and monitor the output. It is not 100% consistent, sometimes it will work for several lines, and then break, sometimes it will break straight away.

### Actual Results
[image]

### Expected Results
[image]

SinglWolf, 2020-11-03:
I confirm. In Windows 7, color is completely absent!
My platformio.ini:
```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = main
lib_dir = components
description = ESP32-Radiola

[env:ESP32Radiola-release]
build_type = release
platform = https://github.com/platformio/platform-espressif32.git
board = esp-wrover-kit
framework = espidf
monitor_speed = 115200
monitor_flags = 
	--echo
	--raw
board_build.partitions = partitions.csv
build_flags = 
	-D FILENAME=ESP32Radiola
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
	-Iinclude
	-Icomponents/audio_player/src
	-Icomponents/tda7313
	-Icomponents/fifo/src
	-Icomponents/ucglib/src
	-Icomponents/xpt2046
extra_scripts = pre:extra_script.py

[env:ESP32Radiola-debug]
build_type = debug
platform = https://github.com/platformio/platform-espressif32.git
monitor_filters = esp32_exception_decoder
board = esp-wrover-kit
framework = espidf
monitor_speed = 115200
monitor_flags = 
	--echo
	--raw
board_build.partitions = partitions.csv
build_flags =
	-D FILENAME=ESP32Radiola-debug
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
	-Iinclude
	-Icomponents/audio_player/src
	-Icomponents/tda7313
	-Icomponents/fifo/src
	-Icomponents/ucglib/src
	-Icomponents/xpt2046
extra_scripts = pre:extra_script.py```

![2020-11-03_07-29-30](https://user-images.githubusercontent.com/32172969/97943632-5606e080-1da6-11eb-8fdd-3fdc3e84dd58.png)

ivankravets, 2022-04-02:
Could you reproduce this issue with the latest PlatformIO Core?

SinglWolf, 2022-04-03:
> Could you reproduce this issue with the latest PlatformIO Core?

I no longer use PlatformIO.

ivankravets, 2022-04-03:
Please reopen if you still need help.

closed 2022-04-03 by ivankravets

45gfg9, 2022-12-19:
Here's running PlatformIO Core 6.1.5 on macOS 13.1. The problem still exists. I suspect Windows terminal color codes just don't work the same way as Unix systems. But it's not working even on Unix systems (at least for my macOS).