platformio/platformio-core #3551: platformio.ini: `platform` directive is not carried over from extended environment

TyIsI, 2020-06-08:
What kind of issue is this?

- [* ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:

Windows 10

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.3.5a2

### Description of problem

platformio.ini: `platform` directive is not carried over from extended environment

#### Steps to Reproduce

1. `pio run -e bluepill1`

### Actual Results

```
> pio run -e bluepill1
Processing bluepill1 (framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Please specify platform for 'bluepill1' environment
```

### Expected Results

Compile

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env]
framework = arduino

[env:bluepill]
platform = ststm32
board = bluepill_f103c8
board_build.core = maple
build_flags = -D ENABLE_USB_SERIAL -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC

[env:bluepill1]
extends = bluepill
build_flags = -D MODULE=bluepill1
```

**Source file to reproduce issue:**
```
#include <Arduino.h>

void setup()
{
    // Do nothing
}

void loop()
{
    // Do nothing
}
```

### Additional info

I've tested this both with my original source code and with a completely untouched version with the above empty source file.

ivankravets, 2020-06-09:
See docs https://docs.platformio.org/en/latest/projectconf/section_env_advanced.html

It should be `extends = env:bluepill`

closed 2020-06-09 by ivankravets

TyIsI, 2020-07-08:
You are right. My bad. Can confirm this works as intended. Thanks for all the great work!