platformio/platformio-core #3554: Use of PLATFORMIO_PENV_DIR does not affect python location (or all tools ?)

bradanlane, 2020-06-09:
What kind of issue is this?

- [X] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**:
Raspberry Pi Zero W running Raspbian Buster

**PlatformIO Version** (`platformio --version`):
PlatformIO Core 4.3.4 (command line only environment)

### Description of problem
After installation, changing the core directory and setting `PLATFORMIO_CORE_DIR` seems to work by it still looks for Python (and possibly other tools) in the original installation folder. Adding `PLATFORMIO_PENV_DIR` for the new location of the subfolder has no effect.

#### Steps to Reproduce

1. Install PlatformIO Core as per instructions (assumed `$HOME` is `/home/user` this puts everything in `/home/user/.platformio` folder and sub folders)
1. `cd $HOME`
1. `mv .platformio new_folder_name`
1. `export PLATFORMIO_CORE_DIR="$HOME/new_folder_name"`
1. `export PLATFORMIO_PENV_DIR="$HOME/new_folder_name/penv"`
1. add `"$PLATFORMIO_CORE_DIR/penv/bin"` to `PATH`
1. attempt `platformio --version`

### Actual Results
Error message: `/home/user/new_folder_name/penv/bin/platformio: /home/user/.platformio/penv/bin/python: bad interpreter: No such file or directory`

### Expected Results
Expected to get version of PlatformIO.

### Additional info
It does not appear to make any difference what `PLATFORMIO_PDEV_DIR` is set to. It appears to always be ignored.

*Perhaps I am using `PLATFORMIO_CORE_DIR` and `PLATFORMIO_PDEV_DIR` incorrectly but it appears to following the documentation.*

ivankravets, 2020-06-09:
You have to set custom system enivronemnts before our installer. So, export custom envs, remove `.platformio`.

closed 2020-06-09 by ivankravets

bradanlane, 2020-06-09:
Thanks.

It would be helpful to users (and avoid my type of issue report) to indicate if an environment variable is required *before* installation. The [current documentation](https://docs.platformio.org/en/latest/envvars.html) implies it can be set at anytime.

ivankravets, 2020-06-09:
Thanks! Updated docs https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR