platformio/platformio-core #3557: Wrong interspercing of error message with verbose output in the terminal

tm-playground, 2020-06-11:
### Configuration

**Operating system**: Linux Mint 19.3 Cinnamon (but the same was observed also on Windows 7)

**PlatformIO Version** (`platformio --version`): version 4.3.4

### Description of problem

When there is an error during build and the verbose mode is active (by running "pio run -v"), the resulting error messages are interlaced with the other output (processed commands and their parameters). Sometimes, even incorrect parts of the output are colorized.

See the attached screenshot of terminal window with highlighted command parameters split (and thus hard to read) by the error message.
[image]

ivankravets, 2020-06-13:
PlatformIO runs a build process in multiple processes. You can run with 1 job using `pio run -j 1`.

closed 2020-06-13 by ivankravets

tm-playground, 2020-06-15:
Ok, I see. But still, since it is an output targeted for a human, I would expect it will be human-readable as much as possible (despite the technical difficulties that the multi-process build introduces).