platformio/platformio-core #3561: Library Dependency Finder includes commented out libraries

positron96, 2020-06-15:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** : PlatformIO, version 4.3.4 + VS Code 1.46.0

### Description of problem

I've found what I think is a bug. If I comment out an `#include` with library in one-line block comment (e.g. `/*#include <EEPROM.h>*/`), the library gets removed from Dependecy graph. This is correct behavior.

However, if I comment the `#include` with a multiline block comment like this: 
```
/*
 #include <EEPROM>
*/
```
then the library still ends up in dependency graph, gets compiled etc.

#### Steps to Reproduce

1. Create a simple project. I used an Arduino framework for Arduino Uno board
2. Try the following code in main.cpp:
```
#include <Arduino.h>

/* #include <EEPROM.h> */

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

3. Build. The output includes lines 
```
Scanning dependencies...
No dependencies
```

4. Now try another code (only 2 newlines added):  
```
#include <Arduino.h>

/* 
#include <EEPROM.h> 
*/

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

5. Build. This time the library gets included in build:
```
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 2.0
```

### Expected Results

In both cases the library should not be included in build.

### Additional info

I'm using PIO from VS Code and installed all stuff via VS Code.

positron96, 2020-06-15:
I've also tried enabling "chain+" mode in `platformio.ini`, but this did not change anything

TD-er, 2020-06-25:
Could it be similar behavior when the include is 'disabled' via #ifdef wrappers?

For example:
```c++
#ifdef USE_LIB_FOO
#include <foo>
#else
#include <bar>
#endif
```

positron96, 2020-06-26:
AFAIK, since default search mode is `chain`, ifdefs are not supposed to work anyway (only in "chain+" or "deep+" modes). Comments, however, are.

Misiu, 2020-07-02:
@TD-er not sure if this help, but I created a quick test and I've added `lib_ldf_mode = chain+` to platform.ini.

main.cpp:
```
#include <Arduino.h>

#ifdef USE_LITTLEFS
#include <FS.h>
#include <LittleFS.h>
#else
#include <FS.h>
#endif

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

platform.ini file:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
build_flags = -D USE_LITTLEFS
lib_ldf_mode = chain+
```

result:
[image]

platform.ini:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
build_flags = -D NOT_USE_LITTLEFS; this has changed
lib_ldf_mode = chain+
```
[image]

I hope this helps 🙂

TD-er, 2020-07-02:
Just for reference: [docs.platformio.org - ldf-mode](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode)

I will try to use the chain+ mode.
But in the past I have also looked into this and then the chain+ mode (or the deep(+) mode) had some issues. Sadly I don't remember what those were.

ivankravets, 2022-04-02:
A temporary solution is to comment each line with `//`. This issue will solved by https://github.com/platformio/platformio-core/issues/3113

closed 2022-04-02 by ivankravets