platformio/platformio-core #3573: Feature request: list custom targets via --list-targets

mcspr, 2020-06-23:
After https://github.com/platformio/platformio-core/issues/3544 it is possible to list IDE targets, but not the custom ones that come from user scripts.

For example, https://github.com/xoseperez/espurna/blob/dev/code/scripts/pio_main.py defines a new `release` target via env.Alias() call:
```python3
# handle `-t release` when CI does a tagged build
env.AlwaysBuild(env.Alias("release", "${BUILD_DIR}/${PROGNAME}.bin", copy_release))
```

However, --list-targets only lists PIO built-in results, as there are some targets missing:
```
$ pio run -e esp8266-4m-git-base --list-targets
Environment          Group     Name       Title                 Description
-------------------  --------  ---------  --------------------  -----------------------------------------------------
esp8266-4m-git-base  Advanced  compiledb  Compilation Database  Generate compilation database `compile_commands.json`
esp8266-4m-git-base  Generic   clean      Clean
esp8266-4m-git-base  Platform  upload     Upload
```

Some observations based on the source:
- This specific list is only updated after a call to the `env.AddTarget()`
https://github.com/platformio/platformio-core/blob/5dadb8749e86e97d429927e0c82add4455acea4a/platformio/builder/tools/piotarget.py#L91
https://github.com/platformio/platformio-core/blob/5dadb8749e86e97d429927e0c82add4455acea4a/platformio/builder/tools/piotarget.py#L61
https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html has no mention of the method, so I am not really sure if this is a public method or something internal.
- SCons seems to keep current aliases via 'ans' object (but I am not really sure whether it is local or global across the program):
```python3
from SCons.Node.Alias import default_ans
print('- default_ans')
for x in default_ans.keys():
    print(x)
print('- projenv.ans')
for x in projenv.ans.keys():
    print(x)
print('- env.ans')
for x in env.ans.keys():
    print(x)
```
Which shows a more complete list:
```
checkprogsize
nobuild
buildprog
size
upload
uploadfs
erase
release
```

ivankravets, 2020-06-24:
The API has been changed. Please upgrade to PIO Core 4.4 and use https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-targets

closed 2020-06-24 by ivankravets

mcspr, 2020-06-25:
Thanks! I could've sworn I search the page...

It also covers another use-case very nicely:
```diff
diff --git a/code/scripts/espurna_utils/lwip.py b/code/scripts/espurna_utils/lwip.py
index a742fcd7..2ed1bd6e 100644
--- a/code/scripts/espurna_utils/lwip.py
+++ b/code/scripts/espurna_utils/lwip.py
@@ -48,9 +48,7 @@ def lwip_inject_patcher(env):
         "Rebuilding lwip",
     )

-    patcher = env.Alias("patch-lwip", None, patch_action)
-    builder = env.Alias("build-lwip", patcher, build_action)
+    patcher = env.AddCustomTarget("patch-lwip", None, patch_action, always_build=True)
+    builder = env.AddCustomTarget("build-lwip", patcher, build_action, always_build=True)
     if os.environ.get("ESPURNA_PIO_PATCH_ISSUE_1610"):
         env.Depends("$BUILD_DIR/${PROGNAME}.elf", builder)
-    env.AlwaysBuild(patcher)
-    env.AlwaysBuild(builder)
```
Can't comment on the IDE extension as it is still in development, but CLI --list-targets works as intended!

ivankravets, 2020-06-25:
You can try the latest beta. It's ery stable https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta-help-us-with-testing/14097

We don't plan to have more changes and we don't release PlatformIO IDE extension for VSCode because still wait on release of PIO Core 4.4.

mcspr, 2020-07-02:
I did try it, and it works! Thanks.

Sorry for the delay though. As I never installed the dev extension on the PC I tested with, I never noticed that `pio upgrade --dev` does not work with PIO installed through `pip install --edit <git-cloned-platformio-core>`. I wonder if 'upgrade' cmd know about that method of installation?