platformio/platformio-core #3574: PIO cuts `.a` when `-l:` option used

puzrin, 2020-06-24:
### Configuration

**Operating system**: Ubuntu 18.04LTS

**PlatformIO Version** (`platformio --version`): 4.4.0a3

### Description of problem

If i need to link library without `lib` prefix, via `-l:` option, i have to type double `.a.a` at the name end:

```
-Lhal/stm32g473rc_usb_pd/stm32cube/Middlewares/ST/STM32_USBPD_Library/Core/lib
-l:USBPDCORE_PD3_CONFIG_1_CM4_wc32.a.a
```

#### Steps to Reproduce

- Use real project to play with https://github.com/puzrin/reflow_micro/blob/master/platformio.ini#L134-L135

### Actual Results

Have to add one more `.a` to library name: `-l:USBPDCORE_PD3_CONFIG_1_CM4_wc32.a.a`

### Expected Results

Expect to use full name, as recommended at [SO](https://stackoverflow.com/questions/1305266/how-to-link-to-a-shared-library-without-lib-prefix-in-a-different-directory):

```
-l:USBPDCORE_PD3_CONFIG_1_CM4_wc32.a
```

valeros, 2020-06-26:
It looks like SCons is not aware of this syntax, so it might be a good idea to raise an issue in [their repository](https://github.com/SCons/scons).
@puzrin Is there any official/reliable source of documentation for this feature besides a post on SO?

puzrin, 2020-06-26:
@valeros i don't know, i use c/cpp or hobby only - almost no experience.

ivankravets, 2020-06-29:
I'm not sure that we can fix this issue. We use SCons as a build system.

@puzrin what is a reason that you can't use `lib` prefix for the archive?

puzrin, 2020-06-29:
See https://github.com/puzrin/reflow_micro/tree/master/hal/stm32g473rc_usb_pd/stm32cube

It's auto-generated by CubeMX (usb pd middleware driver). I keep hal in separate folder, and don't touch there anything by hand. This simplifies maintenance when something should be changed.

PS. I don't use built-in hal because need more fresh and more complete version with mw.

puzrin, 2020-07-02:
I'd suggest to remove `.a` magic as non standard... But, for compatibility, you could apply it only or `-l`, excluding `-l:`.

ivankravets, 2020-07-03:
Please file an issue at https://github.com/SCons/scons/issues and make reference to this issue. Thanks!

closed 2020-07-03 by ivankravets

puzrin, 2020-07-03:
@ivankravets you asked about official docs of `:`. From ld man https://linux.die.net/man/1/ld:

> If namespec is of the form :filename, ld will search the library path for a file called filename, otherwise it will search the library path for a file called libnamespec.a.

--- 

> Please file an issue at [SCons/scons/issues](https://github.com/SCons/scons/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc) and make reference to this issue. Thanks!

Cold you do it yourself? I don't know how to explain that problem is on their side & create a minimal sample.

ivankravets, 2020-07-03:
We don't manipulate with library names or strip them. Please file an issue at https://github.com/SCons/scons/issues.
We will discuss together with SCons Team what to do in this case.