platformio/platformio-core #3578: Endless loop / Memory leak in ccpcheck in a project that uses ArduinoJson

arduhe, 2020-07-04:
- [x ] **PlatformIO Core**.

### Configuration

**Operating system**:
OpenSuse Leap 15.1

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 4.3.4

### Description of problem

I've started _pio check_ from clion - a few seconds later my computer was frozen and I had to switch the power off to restart it.
Further investigations showed me that the problem is located in _pio check_ regardless how you start it. 
I've created a small project to reproduce the problem. You find it at https://github.com/arduhe/cppcheck_fault.git

#### Steps to Reproduce

1.  check out the example project
2. call _pio check_ 

### Actual Results
One core (running cppcheck) shows permanently 100% load and the process cppcheck increases memory usage until the complete available RAM and swap-space are used, unless you kill the process. Otherwise the computer freezes and it's time for a forced reboot 😉

### Expected Results
cppcheck should show the results of it's checks and not freeze my PC by swallowing the whole memory

valeros, 2020-08-25:
Hi @arduhe ! We've updated the Cppcheck package to v2.1 and I'm not able to reproduce the problem, it would be great if you could retest your project with PlatformIO Core from the dev branch. Just run `pio upgrade --dev` in the terminal and you ready to go.

arduhe, 2020-08-26:
After upgrade to latest dev 
`pio --version
PlatformIO, version 5.0.0b2`

pio check terminates properly :-)

Is there a stable version containing this fix? To be honest - I am not a friend of dev versions  - aka beta - no matter from which program ;-).

valeros, 2020-08-26:
Closing as resolved. At the moment there is no stable version with this fix. I think it's safe to use the current beta since it will be released as stable v5.0 very soon.

closed 2020-08-26 by valeros

arduhe, 2020-08-26:
> I think it's safe to use the current beta

That's the same I always say about my software ;-)