platformio/platformio-core #3580: Failed Remote Unit Test / Uploading using Travis : Could not find LD script.

HamzaHajeir, 2020-07-06:
### Configuration

**PlatformIO Version** (`platformio --version`):  4.3.4

### Description of problem

I’m testing remote CI using Travis CI.

I encounter problems testing on my ESP8266 NodeMCU which is attached to my PC (for demo).

What I've tried ? 
[I’ve tried](https://github.com/platformio/platform-espressif8266/issues/165#issuecomment-527202681) `pio remote update` from Travis CI and `pio update` locally. but nothing has changed.

#### Steps to Reproduce

1. Correctly setup with travis-ci.org. So any git push triggers a remote test (See .travis.yml below)
2. Connect NodeMCU to PC and run `pio remote agent start -n mydev`
3. hit `git push` to github.
4. wait for the error.

### Actual Results

This pops up when it tries to reach connected NodeMCU.

> Error: Could not find LD script

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]

platform = espressif8266

board = nodemcuv2

framework = arduino

build_flags = 

  -D BEARSSL_SSL_BASIC

; lib_deps = Blynk

board_build.ldscript = eagle.flash.4m1m.ld

monitor_speed = 115200

upload_speed = 921600

test_ignore = test-native

[env:native]

platform = native

test_ignore = test-embedded
```

**.travis.yml :**
```yml
language: python

python:

    - "3.8"

sudo: false

cache:

    directories:

        - "~/.platformio"

install:

    - pip install -U platformio

    - platformio update

script:

    - pio remote update

    - platformio test -e native

    - pio remote agent list

    - pio remote --agent mydev test -e nodemcuv2

    - pio remote --agent mydev run -e nodemcuv2 -t upload
```

### Additional info
I’ve received the corresponding commands at pio agent :

> 2020-07-03 21:35:07 [info] Remote command received: psync
> 2020-07-03 21:35:12 [info] Remote command received: test
> 2020-07-03 21:35:31 [info] Remote command received: psync
> 2020-07-03 21:35:36 [info] Remote command received: run

What could be the problem / solution ?

HamzaHajeir, 2020-08-09:
Still failing with same error.

Newer Travis-ci log :
https://api.travis-ci.org/v3/job/704757056/log.txt

ivankravets, 2020-09-13:
Could you run `pio remote run  -e nodemcuv2 --force-remote` before `pio remote --agent mydev test -e nodemcuv2
` as a workaround? Did it help?

HamzaHajeir, 2020-09-14:
Thanks @ivankravets for the hint.

But unfortunately, it didn't work. 

You can check by log : 
https://api.travis-ci.org/v3/job/726993416/log.txt

optiprime, 2021-01-08:
Same problem here. Interestingly the problem **does not occur** for board = esp32dev!

Another workaround (one that avoids forcing the build process to run on the agent computer):
Edit ~/.platformio/platforms/espressif8266/builder/main.py on your agent and comment out the two lines after  the comment line "#remove after PIO Core 3.6 release" (in lines ~263-264)

pfeerick, 2023-11-22:
I just hit this myself after trying out `pio remote` again as I have a ESP8266 `d1_mini` _thing_ running off of a PC so thought I might as well make use of that. 

I can report that commenting out the mentioend two lines changes it from "Error: Could not find LD script" to `[SUCCESS]` (and it does upload). 

https://github.com/platformio/platform-espressif8266/blob/d6e84b9927034a619d8e5b9e059ea5ced9ffb836/builder/main.py#L263-L264