platformio/platformio-core #3581: CLion code insight don't work inside ./test folder

IncPlusPlus, 2020-07-07:
- [*] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Pop!_OS

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.4.0a4

### Description of problem
CLion doesn't think that .cpp files inside the ./test folder are part of the project, noting "This file does not belong to any project target, code insight features might not work properly". 

#### Steps to Reproduce

1. `git clone https://github.com/platformio/platformio-examples`
2. Open `platformio-examples/unit-testing/wiring-blink` in CLion and allow it to create CMakeLists.txt and wait for the automatically started `pio init` command to finish
3. Open ./test/test_main.cpp and note the error highlighting on all `#include`s

### Actual Results
Using `#include` leads to import errors (only in CLion, project compiles and tests fine with `pio` CLI tool).

### Expected Results
Syntax highlighting works as expected without having red underscores on all `#include`s and most function calls.

### Additional info

**System information** (`uname -a`): `Linux pop-os 5.4.0-7634-generic #38~1592497129~20.04~9a1ea2e-Ubuntu SMP Fri Jun 19 22:43:37 UTC  x86_64 x86_64 x86_64 GNU/Linux`

**CLion version**: CLion 2020.2 EAP, Build #CL-202.6109.21

**PlatformIO CLion plugin version**: 202.6109.21

dumarjo, 2021-06-22:
still doesn't work.

Evelios, 2021-06-23:
I'm having the same issue. One way that partially gets around the problem is to mark the `test` directory as sources root

[image]

dumarjo, 2021-06-23:
I created a new file CMakeListUser.txt with this content
```cmake
# Add the test source files to CMake so that CLion recognises them properly.
# File won't be required once issue here is closed: https://youtrack.jetbrains.com/issue/CPP-21173
FILE(GLOB_RECURSE TEST_LIST
    ${CMAKE_CURRENT_LIST_DIR}/test/*.*
)

list(APPEND SRC_LIST ${TEST_LIST})
```
and now Clion is happy.

j4m3s, 2022-06-03:
There's a bug open on the jetbrains site for the same thing (has been there for some years now - https://youtrack.jetbrains.com/issue/CPP-21173/Integration-of-tests-in-plugin-PlatformIO-for-CLion) .  Would the bug be fixed via here in the PlatformIO core repo or on the jetbrains side? We should close one or other bug report, and given how simple the fix seems to be I'm happy to have a play around and try to create a PR if someone can point me in the right direction?

j4m3s, 2022-06-03:
Thank you for the pointer in the forum on where to look to create a PR to fix this @ivankravets.  Having gone to do it I've found that you actually fixed it in commit ed33652 which went into release 5.2.1!  

I've checked and the `CMakeListsUser.txt` file is no longer required.  Changing the test_dir value in `platformio.ini` and then re-initialising from CLion `Tools > PlatformIO > Re-Init` updates the path in `CMakeListsPrivate.txt` too, so everything works exactly as expected.  

Propose you close this issue :)

ivankravets, 2022-06-03:
Thanks! Fixed in https://github.com/platformio/platformio-core/commit/ed336525343f3825e748349254afff35c3569f59

closed 2022-06-03 by ivankravets