platformio/platformio-core #3582: Error: [API] Method Not Allowed

blinkinglight, 2020-07-07:
What kind of issue is this?

- [ ] **PlatformIO Core**.
     when trying to register lib "platformio lib register https://raw.githubusercontent.com/9glt/arduino-button/master/library.json" get this "Error: [API] Method Not Allowed"

------------------------------------------------------------------

### Configuration

**Operating system**: osx

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.3.4

### Description of problem
Error: [API] Method Not Allowed

#### Steps to Reproduce

1. platformio lib register https://raw.githubusercontent.com/9glt/arduino-button/master/library.json

### Actual Results
Error: [API] Method Not Allowed

dimircea, 2020-07-08:
I have the same problem when trying to register this library: https://raw.githubusercontent.com/dimircea/REncoder/master/library.json

It appears to be a more general problem...

Regards,
Mircea

ivankravets, 2020-07-10:
Fixed. Please use 
```
pio upgrade --dev
pio package publish --help
```

closed 2020-07-10 by ivankravets