platformio/platformio-core #3584: cannot compile on Mac OS X 10.12.6

mikesmam, 2020-07-08:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

mikesmam, 2020-07-08:
> Executing task: platformio run --environment LPC1768 <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing 0.1.2
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/command.py", line 138, in cli
    is_test_running,
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/processor.py", line 75, in process
    skip_default_package=True,
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py", line 325, in platform_install
    force=force,
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 84, in install
    self, name, requirements, silent=silent, force=force
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 702, in install
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 480, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 195, in download
    raise e
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

ivankravets, 2020-08-17:
1. Please install the latest Python 3 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter
2. Remove `/Users/mikesam/.platformio/penv` folder and restart VSCode

Everything should work now.

closed 2020-08-17 by ivankravets

mikesmam, 2020-08-17:
1. I installed the latest Python 3 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter <https://docs.platformio.org/en/latest/faq.html#install-python-interpreter>
2. could not find the file: Remove /Users/mikesam/.platformio/penv folder and restart VSCode

tried to build ::

> Executing task: platformio run <

Processing STM32F103RE_btt (platform: ststm32; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 7.0.0 > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200602 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
OSError: [Errno 2] No such file or directory: '/Users/mikesam/MarlinFirmware/.pio/libdeps/STM32F103RE_btt/_tmp_installing-2N9ngk-package':
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/Users/mikesam/.platformio/platforms/ststm32/builder/main.py", line 98:
    target_elf = env.BuildProgram()
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 1036:
    project.install_dependencies()
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 852:
    if _is_builtin(uri):
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 844:
    for lb in self.env.GetLibBuilders():
  File "/Users/mikesam/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 963:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 59:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 135:
    self.process_extra_options()
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 266:
    with fs.cd(self.path):
  File "/Users/mikesam/.platformio/penv/lib/python2.7/site-packages/platformio/fs.py", line 35:
    os.chdir(self.new_path)
========================================================================== [FAILED] Took 2.23 seconds ==========================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
[... 35 lines cut ...]
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
==================================================================== 1 failed, 0 succeeded in 00:00:02.235 ====================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Michael Sampognaro
Morgan Hill Cellars / Pedrizzetti Winery
1645 San Pedro Avenue
Morgan Hill, Ca 95037

408 779 7389

www.morganhillcellars.com