platformio/platformio-core #3588: Libraries keep installing everytime the Build is run

m-menon, 2020-07-10:
I am using Visual Studio Code with Platformio IDE extension (1.10.0). Since a few hours there is weird issue with VS Code + Platformio IDE. The VS Code version is 1.47.0.

Everytime I run the Build , a few libraries keep getting installed even though they already exist, or have been installed in the previous Build. I also noticed that the folder naming convention inside .pio->libdeps has changed. The (new library)folder no longer has the _IDXXXX suffix to the folder unlike previously installed library folders. I am attaching a few screenshots for now. If there's any further information required, I can quickly post that.

The Platformio Core version is 4.3.4 (Home 3.2.3)

[image]
[image]
[image]

**NOTE: I just noticed that if I install it using library name instead of library Id, the repeated installation does not happen. It only installs the library on the first Build. So in short, the library installation problem is ONLY when using the library Id (in platformio.ini)**

ivankravets, 2020-07-10:
Sorry, this is a known issue. We switched the whole infrastructure to the new package registry which will be announced soon. We will not support more libraries by ID. We will still allow to install them for compatibility reason but updates, etc., will not work.

A temporary workaround is to use the library name. The upcoming PIO Core 4.4 should have support for the owner declaration. For example,
```
lib_deps = 
   ownername/libraryname @ ^1.2.3
```

Library owner names will be published soon near each library.

m-menon, 2020-07-11:
Thanks for confirming. I wasn't very sure if it was my IDE that had messed up for some reason.

I switched to using library names and everything works as expected.

closed 2020-07-11 by m-menon