platformio/platformio-core #3589: pio test fails during build in CI but works locally

lalten, 2020-07-10:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.3.4

### Description of problem
`pio test` works fine locally:
```bash
git clone https://github.com/lalten/aws4_auth.git
cd aws4_auth
pio test
```
But in CI, a header is not found while compiling a library fetched via lib_deps
https://github.com/lalten/aws4_auth/runs/859773691
```
gcc -o .pio/build/native/lib146/mbedtls/library/aes.o -c -Wall -Wextra -Werror -DPLATFORMIO=40304 -DUNIT_TEST -DUNITY_INCLUDE_CONFIG_H -I.pio/libdeps/native/mbedtls -I.pio/build/native/UnityTestLib -I/home/runner/.platformio/packages/tool-unity .pio/libdeps/native/mbedtls/library/aes.c
.pio/libdeps/native/mbedtls/library/aes.c:29:10: fatal error: mbedtls/config.h: No such file or directory
 #include "mbedtls/config.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/native/lib146/mbedtls/library/aes.o] Error 1
src/Sha256.cpp:5:10: fatal error: mbedtls/md.h: No such file or directory
 #include <mbedtls/md.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/native/src/Sha256.o] Error 1
```

I suspect that the build fails because the mbedtls library does not provide a proper library.json that declares its `include` directory. But then, how does it work locally?

**The content of `platformio.ini`:**
https://github.com/lalten/aws4_auth/blob/feature/ci/platformio.ini
```ini
[platformio]
default_envs = native

[env]
build_flags =
    -Wall
    -Wextra
    -Werror
lib_deps =
    https://github.com/ETLCPP/etl.git#18.4.0
    https://github.com/ARMmbed/mbedtls.git#mbedtls-2.16.6

[env:native]
platform = native
test_build_project_src = yes
```

**GitHub Actions config:**
https://github.com/lalten/aws4_auth/blob/feature/ci/.github/workflows/main.yml
```yaml
name: PlatformIO CI

on: [push, pull_request]

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
    - uses: actions/checkout@v2
      
    - name: Set up Python
      uses: actions/setup-python@v2

    - name: Configure git
      run: git config --global advice.detachedHead false
      
    - name: Install PlatformIO
      run: |
        python -m pip install --upgrade pip
        pip install platformio
      
    - name: Run PlatformIO unit tests
      run: platformio test -v
```

lalten, 2020-07-11:
The issue was indeed that mbedtls is missing a library.json. Platform.io was just using the system-mbedtls installed in `/usr`. I have opened https://github.com/ARMmbed/mbedtls/pull/3490 to add a library.json upstream.

ivankravets, 2020-07-11:
Thanks for the PR! https://github.com/ARMmbed/mbedtls/pull/3490/files 

"version" field is incorrect. It should be `"version": "2.23.0"`

lalten, 2020-07-11:
OK, I assumed that it should mirror the release tags because of the comment on the `repository`/`branch` being ignored in that case.

Actually it would be nice if the version was optional, because then the mbedtls maintainers wouldn't have to update the manifest for every release. Would the LDF / Library Registry still accept the library if the `version` field was missing?

ivankravets, 2020-07-12:
You can actually maintain this library in the PlatformIO registry. Download it, add `'library.json` and run `pio package publish`. You will need the latet PIO Core 4.4-dev (`pio upgrade --dev` to upgrade).

ivankravets, 2020-08-17:
MbedTLS has been added to the PlatformIO Registry https://platformio.org/lib/show/10874/mbedtls

closed 2020-08-17 by ivankravets