platformio/platformio-core #3597: [PlatformIO Core] PlatformIO Home fails to run due to Unexpected Error (missing termios module)

koscum, 2020-07-14:
What kind of issue is this?
**PlatformIO Core**

### Configuration

**Operating system**:
Windows 10, MSYS2 + MinGW64

**PlatformIO Version** (`platformio --version`):
4.3.4

### Description of problem
Running platformio home fails with Unhandled Error due to missing termios module.

#### Steps to Reproduce

1. Install pip under msys2/mingw64 environment
2. Install platformio
3. Manually install contrib-pysite and contrib-piohome (CorePackageManager fails to install them)
4. Run platformio home

### Actual Results
PlatformIO Home stops with unhandled error.

### Expected Results
PlatformIO Home runs normally.

### Additional info
```
Unhandled Error
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:/msys64/mingw64/lib/python3.8/site-packages/platformio/commands/home/command.py", line 132, in cli
    reactor.run()
  File "C:/Users/koscum/.platformio/packages/contrib-pysite/twisted/internet/base.py", line 1283, in run
    self.mainLoop()
  File "C:/Users/koscum/.platformio/packages/contrib-pysite/twisted/internet/base.py", line 1292, in mainLoop
    self.runUntilCurrent()
--- <exception caught here> ---
  File "C:/Users/koscum/.platformio/packages/contrib-pysite/twisted/internet/base.py", line 913, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "C:/msys64/mingw64/lib/python3.8/site-packages/platformio/commands/home/command.py", line 101, in <lambda>
    reactor.callLater(1, lambda: click.launch(home_url))
  File "C:/msys64/mingw64/lib/python3.8/site-packages/click/termui.py", line 613, in launch
    from ._termui_impl import open_url
  File "C:/msys64/mingw64/lib/python3.8/site-packages/click/_termui_impl.py", line 626, in <module>
    import tty
  File "C:/msys64/mingw64/lib/python3.8/tty.py", line 5, in <module>
    from termios import *
builtins.ModuleNotFoundError: No module named 'termios'
```

ivankravets, 2020-08-17:
We don't support the Cygwin environment. It has limited Python interpreter. There even a checker for this https://github.com/platformio/platformio-core/blob/develop/platformio/__main__.py#L74

Please use official installer https://github.com/platformio/platformio-core-installer

closed 2020-08-17 by ivankravets

ustsv, 2021-03-05:
PlatformIO 5.0.4 full work with msys2/mingw64 python 3.8.8, but need patch Click https://github.com/pallets/click/pull/1666,
PlatformIO 5.1.0 SCons.Platform.TempFileMunge is broken, without quotes for a long command line in \platformio\builder\tools\piomaxlen.py
<pre>--------------------------  -----------------------------------
PlatformIO Core             5.0.4
Python                      3.8.8-final.0
System Type                 windows_amd64
Platform                    Windows-7
File System Encoding        utf-8
Locale Encoding             cp1251
PlatformIO Core Directory   C:/Users/name/.platformio
PlatformIO Core Executable  C:\msys2\mingw64\bin\platformio.exe
Python Executable           C:/msys2/mingw64/bin/python.exe
Global Libraries            12
Development Platforms       6
Tools & Toolchains          16
--------------------------  -----------------------------------</pre>
<pre>avr-g++ @Z:/tmp/Marlin/.pio/build/mega2560/tmp9taerze3.tmp
Invalid switch - "tmp".</pre>