platformio/platformio-core #3603: [API] ConnectionError

hzhh110, 2020-07-20:
[API] ConnectionError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AliyunMqttArduino%22+name%3A%22ArduinoJson%22+name%3A%22NTPClient%22 (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) (incremented retry: max=5, total=1)

hzhh110, 2020-07-20:
[image]

hzhh110, 2020-07-20:
APIRequestError: [API] Could not connect to PlatformIO API Service. Please try later.:
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/builder/main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/huangzhenhui/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/huangzhenhui/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/huangzhenhui/.platformio/platforms/espressif32/builder/main.py", line 213:
    target_elf = env.BuildProgram()
  File "/Users/huangzhenhui/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/Users/huangzhenhui/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/huangzhenhui/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 1036:
    project.install_dependencies()
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 868:
    lm.install(uri)
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/managers/lib.py", line 272:
    interactive=interactive,
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/managers/lib.py", line 165:
    "/v2/lib/search", dict(query=" ".join(query)), cache_valid="1h"
  File "/Users/huangzhenhui/.platformio/penv/lib/python3.7/site-packages/platformio/util.py", line 361:
    "Could not connect to PlatformIO API Service. Please try later."

ivankravets, 2020-07-20:
Duplicate of https://github.com/platformio/platformio-core/issues/3604

closed 2020-07-20 by ivankravets