platformio/platformio-core #3613: UnicodeDecodeError: 'ascii' codec can't decode byte

GunZi200, 2020-07-31:
When I try to build my ATmega2560 project I get this on my Desktop PC. But it works fine on my laptop. Both using Windows 10 and same version og PlatformIO.

I am using PlatformIO 4.4.0a7 on both machines.

[image]

`pio upgrade --dev` and restarting VScode does not work.

GunZi200, 2020-07-31:
I tried downgrading to 4.3.4 and the issue is still there.

GunZi200, 2020-07-31:
Ok I fixed the issue by moving project to a folder within C:/ so not within my User folder. Would be nice not having to worry about this.

ivankravets, 2020-08-17:
1. Please run `pio upgrade --dev`
2. Run `pio system info` and share it here.

Does your account name contain non-ascii chars?

GunZi200, 2020-08-17:
Here you go: 
[image]

Can confirm issue is still present when moving from 4.4.0a7 to 4.4.0b1

ivankravets, 2020-08-17:
Please uninstall Python 2.7 and use the latest Python 3. This is a bug of Python 2

GunZi200, 2020-08-17:
This version of Python comes with PlatformIO, I did not install Python 2.7 manually. I removed the penv folder and re-installed and nothing changed. How should I upgrade Python? I have Python 3 installed on my machine.

ivankravets, 2020-08-17:
Could you repeat commands from https://github.com/platformio/platformio-core/issues/3613#issuecomment-674799237 and post result here? Thanks!

GunZi200, 2020-08-17:
Got this: 
[image]

GunZi200, 2020-08-23:
I get an error: 

[image]