platformio/platformio-core #3616: src_filter exlude file from library not works

saloid, 2020-08-02:
What kind of issue is this?

- [*] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version**: `PlatformIO, version 4.4.0a7`

### Description of problem
By default platformio compiles all source files, even if they are not used. So I should manuale exclude unused files in `src_filter`. This don't works in my case. This file is in dependent library and it throws `#error` if detects that it's not used.

#### Steps to Reproduce

1. Try to compile [my project](https://github.com/giver0/Rocket-logger/tree/platformio_src_filter/Soft/Embedded/Main-program). 
2. Compilation will be ended with error.

### Actual Results

```
.pio\libdeps\genericSTM32F103RC\FatFs\src\option\ccsbcs.c:269:2: error: #error This file is not needed at current configuration. Remove from the project.
  269 | #error This file is not needed at current configuration. Remove from the project.
      |  ^~~~~
*** [.pio\build\genericSTM32F103RC\libab5\FatFs\option\ccsbcs.c.o] Error 1
```

### Expected Results
This file should be excluded from build.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
src_filter = -<**/ccsbcs.c>
```
Full platformio.ini file is inside my project [here](https://github.com/giver0/Rocket-logger/blob/platformio_src_filter/Soft/Embedded/Main-program/platformio.ini)

**Source file to reproduce issue:**
Just try to build [my project](https://github.com/giver0/Rocket-logger/tree/platformio_src_filter/Soft/Embedded/Main-program)

### Additional info
I have added CI, so you can see full build sequence and error [here](https://github.com/giver0/Rocket-logger/runs/937597679?check_suite_focus=true).

saloid, 2020-08-02:
Exclusion don't works even if I specify path relative to project dir:
`src_filter = -<.pio\libdeps\genericSTM32F103RC\FatFs\src\option\ccsbcs.c>` (I have tried with `\` and `/` slashes)
and even full path:
`src_filter = -<j:\projects\FireWay\Rocket-logger\Soft\Embedded\Main-program\.pio\libdeps\genericSTM32F103RC\FatFs\src\option\ccsbcs.c>`

ivankravets, 2021-01-22:
You need https://docs.platformio.org/en/latest/librarymanager/config.html#srcfilter

closed 2021-01-22 by ivankravets