platformio/platformio-core #3625: Add support for Inkplate6 Board

funkstille, 2020-08-13:
https://raw.githubusercontent.com/e-radionicacom/Croduino-Board-Definitions-for-Arduino-IDE/master/package_Croduino_Boards_index.json

https://e-radionica.com/en/blog/add-inkplate-6-to-arduino-ide/

davidzovko, 2020-08-27:
Hey,

it's made already, check here: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/issues/16

Just needs to be implemented.

fibis, 2021-03-24:
But if I see it correct, then is https://github.com/e-radionicacom/Inkplate-Arduino-library/issues/16 only about compiling code and not about providing a board definition for platformio.

davidzovko, 2021-03-24:
Brd def can be found here : https://github.com/e-radionicacom/pio-inkplate6

gubouvier, 2023-05-27:
Would it be possible to add all the SolderedElectronic boards ? (not just Inkplate6, but also Inkplate2, etc.)

https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE

nokel, 2023-10-23:
> Would it be possible to add all the SolderedElectronic boards ? (not just Inkplate6, but also Inkplate2, etc.)
> 
> https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE

I second this question as I have an inkplate 10 and it doesn't look like platformio's officially got board definitions for it...