platformio/platformio-core #3627: Exception in pioplatform.py 4.4.0a8

jr-tactiq, 2020-08-16:
### Configuration

**Operating system**: windows 8.1 enterprise

**PlatformIO Version** (`platformio --version`): 4.4.0a8

### Description of problem

I raised a question (https://community.platformio.org/t/test-error-with-mulitple-ways-to-build-the-target/15412) and was advised to upgrade to 4.4.0 as the issue had been fixed.

This worked fine with my platformio running in a ubuntu 18.04.0 VM.

I also have VScode with platformio installed running under windows 8.1 on the same project (shared filesystem with Windows and the VM).

I upgraded the windows/VScode version with `C:\Users\xxx\.platformio\penv\Scripts\platformio.exe upgrade --dev` then tried to run a test in VS code. This showed the following error for each of my three builds (eval board, native and target).

```
AttributeError: 'NoneType' object has no attribute 'version':
  File "C:\users\jryan\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\jryan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\jryan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\users\jryan\.platformio\platforms\native\builder\main.py", line 43:
    target_bin = env.BuildProgram()
  File "C:\Users\jryan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\jryan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\jryan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\jryan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 104:
    env.PrintConfiguration()
  File "C:\Users\jryan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\jryan\.platformio\penv\lib\site-packages\platformio\builder\tools\pioplatform.py", line 223:
    _get_plaform_data(),
  File "C:\users\jryan\.platformio\penv\lib\site-packages\platformio\builder\tools\pioplatform.py", line 153:
    % (platform.title, pkg_metadata.version or platform.version)
```

I added a print to pioplatform.py and found that the issue was here

```python
    def _get_plaform_data():
        data = [
            "PLATFORM: %s (%s)"
            % (platform.title, pkg_metadata.version or platform.version)
        ]
```

as pkg_metadata is None.

This function has changed from the earlier version which simply has

```python
        data = ["PLATFORM: %s %s" % (platform.title, platform.version)]
```
so this seems to be a new bug introduced in the dev version.

As a workaround I edited the pioplatform.py  to read

```python
    def _get_plaform_data():
        if pkg_metadata:
            version = pkg_metadata.version or platform.version
        else:
            version = platform.version
        data = [
            "PLATFORM: %s (%s)"
            % (platform.title, version)
        ]
```
The test then tried to run in VScode (it failed but I think that is an issue with my code).

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:eval_f373vc]
platform = ststm32
board = eval_f373vc
; framework = cmsis
debug_tool = stlink
build_flags = -DEVAL_BOARD
extra_scripts = pre:cart_pre_hal.py
src_filter = -<*> +<common> +<eval_board>
check_patterns =
    src/eval_board/**/*.c
    src/common/**/*.c
test_filter = eval_board
test_transport = custom

[env:target]
platform = ststm32
board = stm32_board
; framework = cmsis
debug_tool = stlink
build_flags = -DTARGET
extra_scripts = pre:cart_pre_hal.py
src_filter = -<*> +<common> +<target>
check_patterns =
    src/target/**/*.c
    src/common/**/*.c
test_filter = target
test_transport = custom

[env:native]
platform = native
build_flags = -DNATIVE
;src_filter = -<*> +<common> +<native> + <common/something/tests>
;check_patterns =
;    src/native/**/*.c
;    src/common/**/*.c
;test_filter = native
test_transport = native

; following copied from sam's stuff
;[env]
;test_build_project_src = true
;check_tool = cppcheck
;check_flags =
;    cppcheck: --std=c11 --inline-suppr --inconclusive -j4 --addon=scripts/analysis/pio_misra.json
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2020-08-16:
Sorry for the issue. Should be fixed with the latest commit. Please re-run `pio upgrade --dev`.

closed 2020-08-16 by ivankravets

jr-tactiq, 2020-08-16:
Works for me.

Thanks for the very quick response.

Please consider adding protection against pkg_metadata not having a version attribute if that could happen
```python
 pkg_metadata.version if pkg_metadata and hasattr(pkg_metadata, 'version') else platform.version,
```

or the shorter but less readable 

```python
getattr(pkg_metadata, 'version', platform.version)
```

which works for a pkg_metadata of None or pkg_metadata without a version attribute.