platformio/platformio-core #3628: platform_packages = framework-arduinoespressif8266 @ https://github.com/tasmota/Arduino.git#2.7.3.2-tasmota

Jason2866, 2020-08-16:
What kind of issue is this?

- [x] **PlatformIO Core**

```
platform                  = espressif8266@2.6.2
platform_packages         = framework-arduinoespressif8266 @ https://github.com/tasmota/Arduino/releases/download/2.7.4.1/esp8266-2.7.4.1.zip
```
and 
```
platform                  = espressif8266@2.6.2
platform_packages         = framework-arduinoespressif8266 @ https://github.com/tasmota/Arduino.git#2.7.3.2-tasmota
```

results in error
`fatal: protocol 'framework-arduinoespressif8266 @ https' is not supported`

------------------------------------------------------------------

### Configuration

**Operating system**: Windows and Linux

**PlatformIO Version** (`platformio --development`):

### Description of problem

platform_packages does not support a repo as target anymore

ivankravets, 2020-08-17:
Sorry for the issue. Please re-test with `pio upgrade --dev`. We did huge refactoring of PlatformIO Package Management System in the upcoming PIO Core 4.4. This is the first huge upgrade since 2014. See https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst

If you will have any issues with package managment (libs_deps, etc), pleaes report. We are going to release PIO Core 4.4 this month.

closed 2020-08-17 by ivankravets

Jason2866, 2020-08-17:
Thx, for fixing fast.

commit 2459e85c (2020-08-17): Fix a bug with the custom platform packages // Resolve #3628