platformio/platformio-core #3631: `platformio check` returns different results with verbose on and off

ademuri, 2020-08-18:
What kind of issue is this

- [ X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): 4.3.4

### Description of problem

The results from `platformio check` vary depending on whether `-v` (verbose) is specified. In this case, no-verbose says that `cppcheck` failed:

```
$ pio check
Checking esp32doit-devkit-v1 > cppcheck (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Error: cppcheck failed to perform check! Please examine tool output in verbose mode.
============================================ [FAILED] Took 0.76 seconds ============================================

Environment          Tool      Status    Duration
-------------------  --------  --------  ------------
esp32doit-devkit-v1  cppcheck  FAILED    00:00:00.759
======================================= 1 failed, 0 succeeded in 00:00:00.759 =======================================
```

verbose says that it passes, but appears to report a result:

```
$ pio check -v                                                 Checking esp32doit-devkit-v1 > cppcheck (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
<snip>
Platform:Unspecified

"severity=error<&PIO&>message=ISO C++ 2011 standard. This support is currently experimental, and must be \

enabled with the -std=c++11 or -std=gnu++11 compiler options.<&PIO&>file=/home/adam/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/c++0x_warning.h<&PIO&>line=32<&PIO&>column=0<&PIO&>callstack=[/home/adam/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/c++0x_warning.h:32]<&PIO&>cwe=0<&PIO&>id=preprocessorErrorDirective"

No defects found
============================================ [PASSED] Took 0.76 seconds ============================================

Environment          Tool      Status    Duration
-------------------  --------  --------  ------------
esp32doit-devkit-v1  cppcheck  PASSED    00:00:00.761
============================================ 1 succeeded in 00:00:00.761 ============================================
```

I expect that these should return the same result. Additionally, the platform species `--std=gnu++11`, as specified in the error, so I'm not sure why that error is shown.

valeros, 2020-08-25:
Hi @ademuri ! This issue should be fixed in https://github.com/platformio/platformio-core/commit/3e72f098fe6e2c52ba1e1094e2b11dd39a9a02c1 . Could you please upgrade your PlatformIO Core to the latest from the dev branch `pio upgrade --dev` and try again? Thanks!

ademuri, 2020-08-26:
Thanks for the quick fix! It works better than before. I'm seeing a new problem, but I'm not sure if it's "my fault".

`pio check` and `pio check -v` now return the same results for me. However, they're giving a Python error. I think it's because my `lib_deps` includes two "libraries" in the parent directory, included like `../constants`. Here's the error:

```
$ pio check
Checking esp32doit-devkit-v1 > cppcheck (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (1.12.4) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting src.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing ../constants
KeyError: 'versions':
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/builder/main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/adam/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/adam/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/adam/.platformio/platforms/espressif32/builder/main.py", line 223:
    target_elf = env.BuildProgram()
  File "/home/adam/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/home/adam/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/home/adam/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 1055:
    project.install_dependencies()
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 891:
    lm.install(spec)
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/package/manager/_install.py", line 48:
    pkg = self._install(
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/package/manager/_install.py", line 99:
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/home/adam/.local/lib/python3.8/site-packages/platformio/package/manager/_registry.py", line 86:
    version = self.pick_best_registry_version(package["versions"], spec)
================================================================================================================== [FAILED] Took 0.36 seconds ==================================================================================================================
```

If I move the "libraries" into the `lib` directory (and remove them from `lib_deps`), everything works as expected. Is this a bug? Should I file a separate issue?

For reference, [here](https://github.com/ademuri/smart-hue-control/tree/master/motion-sensor) is the project.

ivankravets, 2020-08-26:
@ademuri do you use the latest PIO Core dev version? Please re-run `pio upgrade --dev`.

The latest version should say that there is no "../constants" package in the registry. Also, please check the docs how to use local folders for `lib_deps` https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html

closed 2020-08-26 by ivankravets

BlueAndi, 2020-09-05:
Unfortunately it seems I still have this problem:
https://travis-ci.org/github/BlueAndi/esp-rgb-led-matrix/jobs/724327813#L633