platformio/platformio-core #3633: Auto update to pio remote start causing grief

mikesowter, 2020-08-19:
there are almost daily updates to the remote agent on my RPi, and always require manual intervention to correct.
today I can't fix it, whereas normally i waste time figuring out what's wrong with the latest uncalled for and poorly tested technical spam.
So firstly, please advise how to stop these rubbish updates.
secondly, fix this new problem:

    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.7
    creating build/lib.linux-armv7l-3.7/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/model.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/verifier.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/cparser.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/api.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/commontypes.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/cffi_opcode.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/lock.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/recompiler.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/error.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/__init__.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/_embedding.h -> build/lib.linux-armv7l-3.7/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-3.7/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-armv7l-3.7
    creating build/temp.linux-armv7l-3.7/c
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/home/pi/.platformio/penv/include -I/usr/include/python3.7m -c c/_cffi_backend.c -o build/temp.linux-armv7l-3.7/c/_cffi_backend.o
    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
              ^~~~~~~
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.platformio/penv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-iavb43bp/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-iavb43bp/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-y8nv3v2e/install-record.txt --single-version-externally-managed --home /tmp/pip-target-w5ewk0ar --no-compile --install-headers include/site/python3.7/cffi Check the logs for full command output.
Error: Traceback (most recent call last):
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/managers/core.py", line 118, in inject_contrib_pysite
    from OpenSSL import SSL
ModuleNotFoundError: No module named 'OpenSSL'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1256, in invoke
    Command.invoke(self, ctx)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/command.py", line 47, in cli
    inject_contrib_pysite(verify_openssl=True)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/managers/core.py", line 120, in inject_contrib_pysite
    build_contrib_pysite_deps(get_core_package_dir("contrib-pysite"))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/managers/core.py", line 151, in build_contrib_pysite_deps
    dep,
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-compile', '-t', '/home/pi/.platformio/packages/contrib-pysite', 'autobahn == 20.4.3']' returned non-zero exit status 1.

I have the latest version of PlatformIO using
  `pip install -U platformio` command

mikesowter, 2020-08-19:
I forgot to mention that my PKG_CONFIG_PATH environment variable is empty, and there are no libffi files to point to.

ivankravets, 2020-08-23:
It looks like default binaries are not compatible with your OS. See https://github.com/platformio/platformio-core/issues/3490#issuecomment-631093540

closed 2020-08-23 by ivankravets

mikesowter, 2020-08-24:
Thanks Ivan, that worked fine. We still have the issue of auto-update of binaries causing grief. I am using the current RPi build, and I am sure that many others are using RPis for this very useful functionality, so can the RPi be included in your standard test suite? Otherwise, can we either inhibit auto-update or have the facility to roll back to the last proven build? I still lost 5 days development here.  
Regards Mike