platformio/platformio-core #3636: Fails to install package tool-openocd on Windows

manuelbl, 2020-08-23:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 4.3.4

### Description of problem

PlatformIO is unable to install the package *tool-openocd* on Windows. The package is needed for uploads or for debugging.

#### Steps to Reproduce

1. Delete package from `C:\Users\username\.platformio\packages`
2. Execute an upload (e.g. if *st-link* is specified as *upload_protocol* for an *stm32* project)

### Actual Results

The installation and the upload fails 
```
Warning! Package Mirror: [Errno 22] invalid mode ('wb') or filename: 'C:\\Users\\username\\.platformio\\packages\\_tmp_installing-8dezjz-package\\scripts\\target\\1986??1?.cfg'
```

Also see the error reports in https://community.platformio.org/t/openocd-installation-issue/8199/12 (posts since about Aug 10, 2020).

### Expected Results

The package installs successfully and can be used for upload and debugging.

### Additional info

The problem might depend on the selected locale / language. My Windows machine has a mixture of English (United States) and Switzerland. Other users reports problems with German settings.

The core of the problem likely are the Cyrillic filenames *1986ве1т.cfg* and *к1879xб1я.cfg* in OpenOCD (see https://sourceforge.net/p/openocd/code/ci/master/tree/tcl/target/). The *tar* file format has no robust and standardized way for representing such filenames across all platforms. If the *tar* file is created with UTF-8 encoding, the same encoding must be used for decoding. Otherwise the current encoding is used to decode the filenames, resulting in problems.

On my Mac, there was no problem unpacking the file. So as a workaround, I downloaded the file https://bintray.com/platformio/dl-packages/download_file?file_path=tool-openocd-windows_amd64-2.1000.200628.tar.gz on my Mac, unpacked it, copied to Windows, renamed it to `tool-openocd` and put it in the right place.

TheLukaDragar, 2020-08-24:
downloaded file https://bintray.com/platformio/dl-packages/download_file?file_path=tool-openocd-windows_amd64-2.1000.200628.tar.gz renamed to tool-openocd and placed it in C:\Users\UserName\.platformio\packages\ . now it works

ivankravets, 2020-08-24:
Thanks for the report! We re-uploaded all tool-openocd packages. Please re-run `pio update`.

detmot5, 2020-08-25:
the problem still exists
when I use "pio run -t upload" i get an error
Error: Could not install 'tool-openocd' with version requirements '~2.1000.0' for your system 'windows_x86'.

ivankravets, 2020-08-25:
What is your `pio --version`?

detmot5, 2020-08-25:
4.3.4
pio upgrade  says it's newest version

detmot5, 2020-08-25:
Also when I install openocd manually in packaged folder and i use pio update platformio uinstalls it

ivankravets, 2020-08-25:
Sorry :( It was my typo. Now it has been finally fixed. Please re-run `pio update`.

closed 2020-08-25 by ivankravets

LupinIII, 2020-09-12:
I still get the error with the package tool-openocd-esp32. pio --version is 5.0.1

`Tool Manager: Installing platformio/tool-openocd-esp32 @ ~1.1000.20181026
Unpacking  [######################--------------]   63%
Unpacking...
Tool Manager: Warning! Package Mirror: [Errno 22] invalid mode ('wb') or filename: 'C:\\Users\\username\\.platformio\\.cache\\tmp\\pkg-installing-o9ufoc\\share\\openocd\\scripts\\target\\1986??1?.cfg'
Tool Manager: Looking for another mirror...
.
.
.
.
Unpacking  [######################--------------]   63%
Unpacking...
Tool Manager: Warning! Package Mirror: [Errno 22] invalid mode ('wb') or filename: 'C:\\Users\\username\\.platformio\\.cache\\tmp\\pkg-installing-_ou4es\\share\\openocd\\scripts\\target\\1986??1?.cfg'
Tool Manager: Looking for another mirror...
Error: Could not install package 'platformio/tool-openocd-esp32 @ ~1.1000.20181026' for 'windows_amd64' system`

ivankravets, 2020-09-14:
@LupinIII fixed `tool-openocd-esp32` package too. Please re-run `pio update`.

LupinIII, 2020-09-14:
> 
> 
> @LupinIII fixed `tool-openocd-esp32` package too. Please re-run `pio update`.

The installation was now succesful.

```
Tool Manager: Installing platformio/tool-openocd-esp32 @ 1.1000.20190709
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-openocd-esp32 @ 1.1000.20190709 has been installed!
```