platformio/platformio-core #3637: Fail to download platform due to random package.json file in submodule

FireDeveloper, 2020-08-24:
### Configuration

**Operating system**: Win10 64bit

**PlatformIO Version** (`platformio --version`):
version 4.4.0b4

### Description of problem
I am trying to use ESP32 IDF from github and the platform installer throws an error while trying to download the framework.
As far as I can understand it is trying to install a package because there is a package.json file in cJson submodule.
Issue is unrelated to actual building any file, it is just failing to download it properly.

#### Steps to Reproduce

1. Change platformio.ini
2. Build

### Actual Results

`
Processing esp32 (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------C:\Users\Fire\.platformio\platforms\atmelavr\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F6B8E688>
C:\Users\Fire\.platformio\platforms\espressif32\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F6B8E688>
C:\Users\Fire\.platformio\platforms\espressif8266\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73D6B08>
C:\Users\Fire\.platformio\platforms\espressif8266@2.3.1\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73D9908>
C:\Users\Fire\.platformio\platforms\nxplpc\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73DC988>
C:\Users\Fire\.platformio\platforms\nxplpc-arduino-lpc176x\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73DFE48>
C:\Users\Fire\.platformio\platforms\nxplpc-arduino-lpc176x@src-de6b279104eee6e886c6740cfbf2debe\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73E8A08>
C:\Users\Fire\.platformio\platforms\ststm32\platform.json
<platformio.package.manifest.parser.PlatformJsonManifestParser object at 0x000002E4F73EB148>
C:\Users\Fire\.platformio\packages\contrib-piohome\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73D9F88>
C:\Users\Fire\.platformio\packages\contrib-pysite\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73D9B88>
C:\Users\Fire\.platformio\packages\framework-arduino-lpc176x\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73D6748>
C:\Users\Fire\.platformio\packages\framework-arduinoespressif32\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73D6748>
C:\Users\Fire\.platformio\packages\framework-arduinoststm32\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73D6F88>
C:\Users\Fire\.platformio\packages\framework-arduinoststm32-maple\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F6B8ED88>
C:\Users\Fire\.platformio\packages\framework-cmsis\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F6B8E808>
C:\Users\Fire\.platformio\packages\framework-espidf\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F6B8E648>
C:\Users\Fire\.platformio\packages\framework-stm32cube\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73EBD88>
C:\Users\Fire\.platformio\packages\tool-avrdude\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73EB448>
C:\Users\Fire\.platformio\packages\tool-cmake\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73EB888>
C:\Users\Fire\.platformio\packages\tool-cppcheck\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73EB448>
C:\Users\Fire\.platformio\packages\tool-dfuutil\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73DE608>
C:\Users\Fire\.platformio\packages\tool-esptool\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73DEA88>
C:\Users\Fire\.platformio\packages\tool-esptoolpy\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73DEA88>
C:\Users\Fire\.platformio\packages\tool-idf\package.json
<platformio.package.manifest.parser.PackageJsonManifestParser object at 0x000002E4F73F85C8>
C:\Users\Fire\.platformio\packages\tool-jlink\package.json
[... 344 lines cut ...]

The terminal process "C:\Users\Fire\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = esp32 ;ESP32 IDF Enviroment

[env:esp32]
platform = espressif32
board = esp32dev
framework = espidf
platform_packages = framework-espidf @ https://github.com/espressif/esp-idf.git
```

### Additional info
I added a print in parser.py to trace the error. Unchanged core won't print information about the file that it is trying to parse.
I corrected line numbers on output log to match stock parser.py file

ivankravets, 2020-08-24:
Thanks for the report! Please re-test with `pio upgrade --dev`.

FireDeveloper, 2020-08-26:
It didn't work. Failed at the same spot.
PlatformIO version 5.0.0b2 

```
Processing esp32 (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing git+https://github.com/espressif/esp-idf.git
git version 2.28.0.windows.1
Cloning into 'C:\Users\Fire\.platformio\.cache\tmp\pkg-installing-jmgi3e6e'...
remote: Enumerating objects: 8196, done.
remote: Counting objects: 100% (8196/8196), done.
remote: Compressing objects: 100% (6975/6975), done.
remote: Total 8196 (delta 1792), reused 3531 (delta 891), pack-reused 0R
Receiving objects: 100% (8196/8196), 38.44 MiB | 1.34 MiB/s, done.
Resolving deltas: 100% (1792/1792), done.
Updating files: 100% (7164/7164), done.
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
Submodule 'components/cbor/tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'components/cbor/tinycbor'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/tinyusb/tinyusb' (https://github.com/espressif/tinyusb.git) registered for path 'components/tinyusb/tinyusb'
Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Submodule 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' (https://github.com/espressif/esp-cryptoauthlib.git) registered for path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-jmgi3e6e/components/asio/asio'...
remote: Enumerating objects: 10, done.        
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 39225 (delta 0), reused 10 (delta 0), pack-reused 39215        
Receiving objects: 100% (39225/39225), 13.81 MiB | 1.33 MiB/s, done.
Resolving deltas: 100% (26421/26421), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-jmgi3e6e/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
remote: Enumerating objects: 765, done.        
remote: Total 765 (delta 0), reused 0 (delta 0), pack-reused 765        
Receiving objects: 100% (765/765), 511.80 KiB | 1.15 MiB/s, done.
Resolving deltas: 100% (464/464), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-jmgi3e6e/components/bt/controller/lib'...
remote: Enumerating objects: 165, done.        
remote: Counting objects: 100% (165/165), done.        
remote: Compressing objects: 100% (61/61), done.        
remote: Total 549 (delta 145), reused 110 (delta 104), pack-reused 384        
Receiving objects: 100% (549/549), 2.47 MiB | 1.31 MiB/s, done.
Resolving deltas: 100% (349/349), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-jmgi3e6e/components/bt/host/nimble/nimble'...
remote: Enumerating objects: 14, done.        
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (14/14), done.        
remote: Total 31556 (delta 1), reused 4 (delta 0), pack-reused 31542        
[... cut ...]
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "C:\Users\Fire\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

ivankravets, 2020-08-26:
Thanks for the report! Please re-run `pio upgrade --dev`. Should work now.

FireDeveloper, 2020-08-29:
Now the download works (kind of) but the files are missing from packages folder.
It copied only the files from random package.json git
https://github.com/DaveGamble/cJSON/tree/master/tests/json-patch-tests

```
Processing esp32 (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing git+https://github.com/espressif/esp-idf.git
git version 2.28.0.windows.1
Cloning into 'C:\Users\Fire\.platformio\.cache\tmp\pkg-installing-y5pkakez'...
remote: Enumerating objects: 8196, done.
remote: Counting objects: 100% (8196/8196), done.
remote: Compressing objects: 100% (6973/6973), done.
remote: Total 8196 (delta 1793), reused 3536 (delta 893), pack-reused 0 eceiving objects: 100% (8196/8196), 38.07 MiB | 1.33 MiB/s
Receiving objects: 100% (8196/8196), 38.44 MiB | 1.30 MiB/s, done.
Resolving deltas: 100% (1793/1793), done.
Updating files: 100% (7164/7164), done.
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
Submodule 'components/cbor/tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'components/cbor/tinycbor'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/tinyusb/tinyusb' (https://github.com/espressif/tinyusb.git) registered for path 'components/tinyusb/tinyusb'
Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Submodule 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' (https://github.com/espressif/esp-cryptoauthlib.git) registered for path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-y5pkakez/components/asio/asio'...
remote: Enumerating objects: 10, done.        
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 39225 (delta 0), reused 10 (delta 0), pack-reused 39215        
Receiving objects: 100% (39225/39225), 13.81 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (26421/26421), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-y5pkakez/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
remote: Enumerating objects: 765, done.        
remote: Total 765 (delta 0), reused 0 (delta 0), pack-reused 765        
Receiving objects: 100% (765/765), 511.80 KiB | 1.08 MiB/s, done.
Resolving deltas: 100% (464/464), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-y5pkakez/components/bt/controller/lib'...
remote: Enumerating objects: 168, done.        
remote: Counting objects: 100% (168/168), done.        
remote: Compressing objects: 100% (64/64), done.        
remote: Total 552 (delta 147), reused 111 (delta 104), pack-reused 384        
Receiving objects: 100% (552/552), 2.47 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (351/351), done.
Cloning into 'C:/Users/Fire/.platformio/.cache/tmp/pkg-installing-y5pkakez/components/bt/host/nimble/nimble'...
[... cut ...]
=================================================================================================== 1 succeeded in 00:06:25.479 =================================================================================================== 

Terminal will be reused by tasks, press any key to close it.
```

framework-espidf folder contains the files:
.editorconfig
.gitignore
.npmignore
.piopm
cjson-utils-tests.json
package.json
README.md
spec_tests.json
tests.json

Which are exactly the same with here https://github.com/DaveGamble/cJSON/tree/master/tests/json-patch-tests

FireDeveloper, 2020-09-04:
Issue still exists on version 5.0.0

ivankravets, 2020-09-10:
> https://github.com/espressif/esp-idf.git

This is not a valid package, it does not contain `package.json` in the root.

P.S: You can't override ESP-IDF in runtime. We support only stable IDF releases.

commit 13db51a5 (2020-08-24): Install/Uninstall dependencies only for library-type packages // Resolve #3637
commit d5941643 (2020-08-26): Parse npm-like "repository" data from a package manifest // Resolve #3637
closed 2020-09-10 by ivankravets