platformio/platformio-core #3639: Add support for W600-pico board

plesatejvlk, 2020-08-24:
Hi, I'd love to see native support for the W600-pico board: https://www.wemos.cc/en/latest/w600/w600_pico.html

Thanks!

wjbuys, 2020-09-08:
In case it helps, from [this forum thread](https://community.platformio.org/t/w600-pico-support/15516), it looks like there is already an [Arduino core](https://github.com/w600/arduino), and [a WIP implementation of the W60x platform](https://github.com/maxgerhardt/platform-w60x).

maxgerhardt, 2020-09-08:
Will also be extended to compile the Arduino core for the W600 series soon.

Jason2866, 2020-09-12:
@maxgerhardt  cool to have the possibility too use the w600 with arduino support with platformio
Maybe you are considering to do a platformio setup for the RTL8710?
https://github.com/ambiot/amb1_arduino
There is a port for the amb1-sdk but not for Arduino :-(
https://github.com/chris-hatton/platformio-realtek-ameba/tree/amb1-sdk

maxgerhardt, 2020-09-12:
@Jason2866 interestingly enough I actually do have such a board and also used the Arduino core (inside Arduino IDE), might have a look at that after that.

Jason2866, 2020-09-12:
Would be realy cool. One of our developers from Tasmota (tasmota.com) would begin to start a port to the RTL8710. Many new bought devices (smart switches, bulbs) uses the RTL8710.
The ESP8266 is dying in commercial devices

maxgerhardt, 2020-09-13:
Update: https://github.com/maxgerhardt/platform-w60x can now compile the Arduino core for all the same boards the original Arduino core can. Upload via serial works (bootloader upload with PA0 -> GND, RESET)

[image]

[image]

Currently only works for Windows because I'm relying on compiler and tools packages which I have only compiled / published for Windows yet. Will be cleaned up with added Linux and Mac support soon.

And of course, programming the chip with the latest vendor-provided SDK (WM_SDK_W60X_G3.04.00_2003) still works too: 

[image]

wjbuys, 2020-09-24:
Fantastic work @maxgerhardt! I tried out https://github.com/maxgerhardt/platform-w60x (wanted to contribute the Wemos W600 Pico board definition), but it looks like the `tool-openocd-w60x` package isn't published anywhere:

```
Processing w600 (platform: C:\Users\Jacob\Projects\platform-w60x; board: wemos_w600_pico; framework: arduino)
------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing tool-openocd-w60x @ ~1.1000.0
Error: Could not find the package with 'tool-openocd-w60x @ ~1.1000.0' requirements for your system 'windows_amd64'
```

Are you using a locally installed version? I'd be happy to help with packaging that if you can point me in the right direction.

Jason2866, 2020-09-24:
@wjbuys  it seems since platformio v.5.0 it doesnt work anymore how it is done.
The tool is reference here https://github.com/maxgerhardt/platform-w60x/blob/master/manifest.json

maxgerhardt, 2020-09-24:
@wjbuys I do work on PIO core 5 but the old platform.json file still references the manifest -- this is just ignored. I've published the missing `tool-openocd-w60x` package which will hopefully be available tomorrow. The platform however doesn't yet correctly use `openocd` to upload the firmware, that's still WIP.  Also the package is Windows only. 

For Linux I still have to compile the `tool-w60x-download` and use a arm-nonee-eabi-gcc that's directly executable on x64 Linux. The one that the Arduino IDE uses is actually 32-bit and will not run without install a 32-bit version of libc / libstd++ package to be able to run the 32-bit executable. 

> wanted to contribute the Wemos W600 Pico board definition

I've also incidentally got like 6 of those and will add them too. What's disappointing me on that board is however no auto-reset into bootloader mode via DTR+RTS going to Reset+PA0. Only DTR goes to reset which allows automated reset, but not into bootloder mode :(. So still manual pushbutton action is needed.

maxgerhardt, 2020-10-25:
Just to update that this is not a dead project: I'm currently fixing up some packages and adding upload via SWD, e.g. an STLink. Some quirks are still to be worked at and straightened out, but full debugging (breakpoints, variable watches, ...) works. 

[image]

choryuidentify, 2021-05-08:
Hi. I'm trying to make use PlatformIO to W600-pico, so trying to use library of @maxgerhardt

but I have some problems to use this....

Firstly, occured toolchain error.
PlatformIO wants 'platformio/toolchain-gccarmnoneeabi @ ~1.40903.0' but it's not exist anywhere.
So manually add 'platform_packages = maxgerhardt/toolchain-gccarmnoneeabi@~1.40903.0' into 'platformio.ini', and build is fine now. (Solved)

Secondly, occured upload error.
```
Processing generic_w600 (platform: maxgerhardt/tool-w60x-download; board: generic_w600; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing maxgerhardt/tool-w60x-download
Error: Could not find the package with 'maxgerhardt/tool-w60x-download' requirements for your system 'windows_amd64'
The terminal process "C:\Users\chory\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'generic_w600'" terminated with exit code: 1.
```

This package looks not exist anywhere, so I've remove that requirements from 'platform-w60x' definition.
And finally get this!
```
Processing generic_w600 (platform: w60x; board: generic_w600; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/w60x/generic_w600.html
PLATFORM: WinnerMicro W60X (6.1.1+sha.69dbe90) > Generic W600 board
HARDWARE: W600 80MHz, 224KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduino-w60x 1.207.20200914 (2.7)
 - tool-w60x-download 1.10000.0 (1.0.0)
 - toolchain-gccarmnoneeabi 1.40903.0 (4.9.3)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Creating images from .pio\build\generic_w600\firmware.bin
Checking size .pio\build\generic_w600\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  59.2% (used 135680 bytes from 229376 bytes)
Flash: [===       ]  33.4% (used 350240 bytes from 1048576 bytes)
generate normal image completed.

compress binary completed.

generate compressed image completed.

generate debug image completed.

generate flash file completed.

Configuring upload protocol...
AVAILABLE: jlink, serial, stlink
CURRENT: upload_protocol = serial
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\generic_w600\wm_w600.fls
connecting serial...

serial connected.

wait serial sync.........

please manually reset the device.

..

serial sync sucess.

mac 28-6D-CD-**-**-**.

start download.

0% [

download failed, please reset and try again.
```

...but, yeah. It's failed.

I'm trying to manual flash the binary by latest wm_tool.exe from W60X SDK, but still no lucks.

How to configure it can works properly?

maxgerhardt, 2021-05-08:
Hm weird, it displays the MAC but fails.. You reset the W600 chip into bootloader mode by connecting PA0 to GND and pressing reset, right? 

I haven't developed my w600 platform further in quite some time, I guess I should look over it again and clean things up.

choryuidentify, 2021-05-08:
> 
> 
> Hm weird, it displays the MAC but fails.. You reset the W600 chip into bootloader mode by connecting PA0 to GND and pressing reset, right?
> 
> I haven't developed my w600 platform further in quite some time, I guess I should look over it again and clean things up.

Hmmmm what's going on?

I'm not connecting PA0 to GND before, because I can upload W600 SDK demo image without that.
Run wm_tool and reset board, Tada! It flashed fine.

But... I've connect that pins and try flashing again, It FLASHED!

Thank you! @maxgerhardt !

maxgerhardt, 2021-05-08:
There's two bootloaders on the device -- the one, low level, in the mask ROM, entered automatically by having PA0 pulled low during a reset, and then a second-stage bootloader that is always flashed with an application. These bootloaders have different capabilities in the firmware formats they accept. There are WinnerMicro document on this -- [this](http://www.winnermicro.com/en/upload/1/editor/1559616939540.pdf) and [this](http://www.winnermicro.com/en/upload/1/editor/1559640550604.pdf) and [this](http://www.winnermicro.com/en/upload/1/editor/1559640549130.pdf). The upload method that I used probably expects the mask-ROM bootloader.