platformio/platformio-core #3641: Many output lines mentioning "IGNORED"

jcw, 2020-08-25:
- [X] **Feature Request**.

The new 4.4 setup is very flexible, but including all boards and using only one is somewhat inconvenient:

1. a very long list of "IGNORED" lines at the end of a run (even more at the end of a test)
2. those lists do not appear to be sorted in any way

Please consider omitting output about things PIO did _not_ do, as default - perhaps with an "N ignored" message.

Marginally related: in silent mode, i.e. `-s` flag, the output is a bit inconsistent for different target. Some report two build sizes (text & data), some one, some none at all. And `pio run -t size -s` tends to output nothing at all.

**PlatformIO Version**: 4.4.0b6

(copied from #3583)

ivankravets, 2020-09-02:
Thanks! Please re-test with `pio upgrade --dev`.

closed 2020-09-02 by ivankravets

jcw, 2020-09-02:
Confirmed! \o/

commit 44c2b653 (2020-09-02): Show ignored project environments only in the verbose mode // Resolve #3641