platformio/platformio-core #3646: Compilation works when using the VSCode IDE button but fails when running from cli

ASL07, 2020-09-02:
What kind of issue is this?
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Os X

**PlatformIO Version** (`platformio --version`): 5.0.0b3

### Description of problem

Compilation works when using the IDE but fails when running from cli

#### Steps to Reproduce

1.  `rm-rf .pio`
2.  `rm -rf .vscode`
3.   I also tried running `pio run` without deleting those two directories and experience the same issue
3.  `pio run --environment myenvironment`
4.   I get `.pio/build/myenvironment/.sconsign37.db.dblite: No such file or directory`
 
**Please note that if I compile by pressing the build the button on the VSCode left tab, compilation works fine, so my platformio.ini is should not be the problem**

ivankravets, 2020-09-02:
Do you have multiple PIO Core in a system? https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system

ivankravets, 2020-09-03:
Please upgrade to the latest PIO Core 5.0

closed 2020-09-03 by ivankravets