platformio/platformio-core #3651: Unit Test Compile-Only

programminghoch10, 2020-09-04:
What kind of issue is this?

- [x] **Feature Request**.

------------------------------------------------------------------

I would like to have a command which only compiles all the unit tests, but does not actually run them.
I know this is possible with `platformio ci` command, but it is a tedious process to setup, as there must be a seperate setup and run of this command for each test.
I initially thought that `platformio test --without-uploading` would just skip the testing, but that still requires a device connected.

So I suggest addition of `platformio test --compile-only` or sth like that to compile all the tests, so that one can see compiler issues and such without actually the need of a device connected.

This would also make it easier to "compile-test" the unit tests in the CI, for people who don't want or can't ArduinoFake it.

valeros, 2020-09-04:
There is also another option `--without-testing`, if you specify both `--without-uploading --without-testing` then PlatformIO should only compile the tests without running them on a target.

closed 2020-09-04 by valeros

programminghoch10, 2020-09-04:
then please add that to the docs, i looked on [this](https://docs.platformio.org/en/latest/core/userguide/cmd_test.html) page and it wasn't there
thanks anyways!

ivankravets, 2020-09-04:
@programminghoch10 thanks, fixed in https://github.com/platformio/platformio-core/commit/39883e8d683805f897d1c9fa66b82bcc87d04888