platformio/platformio-core #3652: pio remote agent throw's exception on raspberry pi's (5.0.0)

fietser28, 2020-09-04:
I have a problem running the remote agent on raspberry Pi's with the new 5.0.0:

### Configuration

**Operating system**: Raspbian 10 on Raspberry Pi 4 and Raspbian 9 on Raspberry Pi Zero (tested on both, raspbian 10 is pretty clean, first time PlatformIO installation).

**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.0

(penv) pi@troel3:~ $ pio system info
--------------------------  ----------------------------------------
PlatformIO Core             5.0.0
Python                      3.7.3-final.0
System Type                 linux_armv7l
Platform                    Linux-5.4.51-v7l+-armv7l-with-glibc2.28
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pi/.platformio
PlatformIO Core Executable  /home/pi/.platformio/penv/bin/platformio
Python Executable           /home/pi/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2
--------------------------  ----------------------------------------

### Description of problem

- pio remote agent start 
- pio remote agent list
both will throw an exeption (see below)

pio remote agent 
produces the general help text. Other pio commands seem to work ok.

#### Steps to Reproduce

1. Do a clean install of platform (wget https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -O get-platformio.py; python3 get-platformio.py)
2. Login with:                 pio account login (works ok)
3. start remote agent:   pio agent remote start

### Actual Results

(penv) pi@troel3:~ $ pio remote agent start                                                                                                           [89/395]
Tool Manager: Installing platformio/contrib-pysite @ ~2.37.0
Unpacking  [####################################]  100%
Tool Manager: contrib-pysite @ 2.37.191020 has been installed!   
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting twisted==20.3.0                            
  Using cached https://www.piwheels.org/simple/twisted/Twisted-20.3.0-cp37-cp37m-linux_armv7l.whl (3.1 MB)                        
Collecting constantly>=15.1
  Using cached constantly-15.1.0-py2.py3-none-any.whl (7.9 kB)
Collecting Automat>=0.3.0
  Using cached Automat-20.2.0-py2.py3-none-any.whl (31 kB)
Collecting PyHamcrest!=1.10.0,>=1.9.0
  Using cached PyHamcrest-2.0.2-py3-none-any.whl (52 kB)
Collecting hyperlink>=17.1.1
  Using cached hyperlink-20.0.1-py2.py3-none-any.whl (48 kB)
Collecting incremental>=16.10.1
  Using cached incremental-17.5.0-py2.py3-none-any.whl (16 kB)
Collecting attrs>=19.2.0
  Using cached attrs-20.1.0-py2.py3-none-any.whl (49 kB)
Collecting zope.interface>=4.4.2
  Using cached https://www.piwheels.org/simple/zope-interface/zope.interface-5.1.0-cp37-cp37m-linux_armv7l.whl (223 kB)
Collecting six
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting idna>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting setuptools
  Using cached setuptools-50.2.0-py3-none-any.whl (784 kB)
Installing collected packages: constantly, attrs, six, Automat, PyHamcrest, idna, hyperlink, incremental, setuptools, zope.interface, twisted
Successfully installed Automat-20.2.0 PyHamcrest-2.0.2 attrs-20.1.0 constantly-15.1.0 hyperlink-20.0.1 idna-2.10 incremental-17.5.0 setuptools-50.2.0 six-1.15
[... 133 lines cut ...]
  `pip install -U platformio` command
                                                                                             
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html                                                             
                                        
* Report this problem to the developers                                                              
  https://github.com/platformio/platformio-core/issues   
                                                                                                                                                              
============================================================

### Expected Results

Not an exception but a running agent. :-)

### Additional info

- Every time I run a pio remote ... command the package installations shown above are executed. Normally I expect this to run only once after an installation/upgrade. I have no explanation why this happens, probably because of the exception?

- Tried several things including install additional libraries and or -dev packages or adding packages manually (pip install ..). No difference at all.

- The method that throws the error is present in the .py file. Doesn't look strange either, but I'm not very experienced with python.

- activating penv or not makes no difference.

Did I still miss something or is this a bug?

fietser28, 2020-09-04:
[dpkg-l.txt](https://github.com/platformio/platformio-core/files/5176609/dpkg-l.txt)

dpkg -l list off mentioned pi4 with raspbian 10.4

liebman, 2020-09-05:
I have the same exact issue on a pi3 :-(
```
(penv) pi@lab:~ $ pio system info
--------------------------  ----------------------------------------
PlatformIO Core             5.0.0
Python                      3.7.3-final.0
System Type                 linux_armv7l
Platform                    Linux-4.19.97-v7+-armv7l-with-glibc2.28
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pi/.platformio
PlatformIO Core Executable  /home/pi/.platformio/penv/bin/platformio
Python Executable           /home/pi/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       0
Tools & Toolchains          1
--------------------------  ----------------------------------------
```

BertvanHoekelen, 2020-09-05:
I also have this issue on the rpi 4 with ubuntu server.
```
--------------------------  ---------------------------------------------
PlatformIO Core             5.0.0
Python                      3.8.2-final.0
System Type                 linux_aarch64
Platform                    Linux-5.4.0-1016-raspi-aarch64-with-glibc2.29
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/ubuntu/.platformio
PlatformIO Core Executable  /home/ubuntu/.platformio/penv/bin/platformio
Python Executable           /home/ubuntu/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2
--------------------------  ---------------------------------------------
```

BertvanHoekelen, 2020-09-05:
I got it working by installing build-essentials:

```sh
sudo apt-get update
sudo apt-get install build-essential
```

BertvanHoekelen, 2020-09-05:
Never mind, it's still not working. Calling `pio remote` now works but when I try `pio remote agent list` or `pio remote agent start` I get another error. 

```
ubuntu@remote-programmer:~$ pio remote agent list
Tool Manager: Installing platformio/contrib-pysite @ ~2.38.0
Unpacking  [####################################]  100%          
Tool Manager: contrib-pysite @ 2.38.191020 has been installed!
Processing ./.cache/pip/wheels/f2/36/1b/99fe6d339e1559e421556c69ad7bc8c869145e86a756c403f4/Twisted-20.3.0-cp38-cp38-linux_aarch64.whl
Collecting PyHamcrest!=1.10.0,>=1.9.0
  Using cached PyHamcrest-2.0.2-py3-none-any.whl (52 kB)
Collecting attrs>=19.2.0
  Using cached attrs-20.2.0-py2.py3-none-any.whl (48 kB)
Collecting hyperlink>=17.1.1
  Using cached hyperlink-20.0.1-py2.py3-none-any.whl (48 kB)
Collecting incremental>=16.10.1
  Using cached incremental-17.5.0-py2.py3-none-any.whl (16 kB)
Collecting constantly>=15.1
  Using cached constantly-15.1.0-py2.py3-none-any.whl (7.9 kB)
Processing ./.cache/pip/wheels/40/60/ff/d92f4e4d4bd8f303bca761194f4a8c3d207dbc092b1cd33721/zope.interface-5.1.0-cp38-cp38-linux_aarch64.whl
Collecting Automat>=0.3.0
  Using cached Automat-20.2.0-py2.py3-none-any.whl (31 kB)
Collecting idna>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting setuptools
  Using cached setuptools-50.3.0-py3-none-any.whl (785 kB)
Collecting six
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Installing collected packages: PyHamcrest, attrs, idna, hyperlink, incremental, constantly, setuptools, zope.interface, six, Automat, twisted
Successfully installed Automat-20.2.0 PyHamcrest-2.0.2 attrs-20.2.0 constantly-15.1.0 hyperlink-20.0.1 idna-2.10 incremental-17.5.0 setuptools-50.3.0 six-1.15.0 twisted-20.3.0 zope.interface-5.1.0
Collecting autobahn==20.4.3
  Using cached autobahn-20.4.3-py2.py3-none-any.whl (1.1 MB)
Collecting txaio>=20.3.1
  Using cached txaio-20.4.1-py2.py3-none-any.whl (30 kB)
Collecting cryptography>=2.7
  Using cached cryptography-3.1-cp35-abi3-manylinux2014_aarch64.whl (2.6 MB)
Collecting six>=1.4.1
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Processing ./.cache/pip/wheels/c9/26/2c/afdca4bd043261256d528c4b0044f074af7212377da5af9913/cffi-1.14.2-cp38-cp38-linux_aarch64.whl
Collecting pycparser
  Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)
Installing collected packages: txaio, six, pycparser, cffi, cryptography, autobahn
Successfully installed autobahn-20.4.3 cffi-1.14.2 cryptography-3.1 pycparser-2.20 six-1.15.0 txaio-20.4.1
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/twisted already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/six.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/six-1.15.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/bin already exists. Specify --upgrade to force replacement.
Collecting json-rpc==1.13.0
  Using cached json_rpc-1.13.0-py2.py3-none-any.whl (41 kB)
Installing collected packages: json-rpc
Successfully installed json-rpc-1.13.0
Collecting pyopenssl>=16.0.0
  Using cached pyOpenSSL-19.1.0-py2.py3-none-any.whl (53 kB)
Collecting cryptography>=2.8
  Using cached cryptography-3.1-cp35-abi3-manylinux2014_aarch64.whl (2.6 MB)
Collecting six>=1.5.2
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Processing ./.cache/pip/wheels/c9/26/2c/afdca4bd043261256d528c4b0044f074af7212377da5af9913/cffi-1.14.2-cp38-cp38-linux_aarch64.whl
Collecting pycparser
  Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)
Installing collected packages: pycparser, cffi, six, cryptography, pyopenssl
Successfully installed cffi-1.14.2 cryptography-3.1 pycparser-2.20 pyopenssl-19.1.0 six-1.15.0
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/pycparser already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/_cffi_backend.cpython-38-aarch64-linux-gnu.so already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/pycparser-2.20.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/six.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/cffi-1.14.2.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/six-1.15.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/cryptography-3.1.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/cryptography already exists. Specify --upgrade to force replacement.
WARNING: Target directory /home/ubuntu/.platformio/packages/contrib-pysite/cffi already exists. Specify --upgrade to force replacement.
[... 69 lines cut ...]
  File "/home/ubuntu/.platformio/packages/contrib-pysite/OpenSSL/crypto.py", line 12, in <module>
    from cryptography import x509
[... cut ...]
ImportError: cannot import name '_get_backend' from 'cryptography.hazmat.backends' (/home/ubuntu/.platformio/packages/contrib-pysite/cryptography/hazmat/backends/__init__.py)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ubuntu@remote-programmer:~$ 
```

j2thex, 2020-09-06:
same.... How can this be fixed?

ivankravets, 2020-09-06:
This is an issue with Cryptography pre-built binaries. I opened an issue https://github.com/pyca/cryptography/issues/5455

In any case, we did a workaround on our side. Could you performa the next commands?

1. `pio upgrade --dev`
2. `pio home`

It should recompile evrything from source. Does it work now?

closed 2020-09-06 by ivankravets

fietser28, 2020-09-06:
It works on my Pi4 with raspbian 10.4. Will start the build on the Pi Zero tonight, rebuilding everything on a Pi Zero is very slow.

The first time I tried 'pio remote agent start' it failed, but all on subsequent attempts it worked. Commands like remote upload and device list also work fine now.

Thanks for the workaround! Keep up this amazing work.

fietser28, 2020-09-07:
It also works on my Pi Zero with Raspbian 9. First time I ran 'pio remote agent start' it failed, second time it worked. 

I also tried to remove the python3-cryptography packages (v1.7.1) and install cryptography thru pip (v3.1), but that didn't work for me (I'm not very proficient in python/pip package management, although I learned a lot the last few days.)

commit 4234dfb6 (2020-09-06): Fixed an issue with "ImportError: cannot import name '_get_backend' from 'cryptography.hazmat.backends'" when using Remote Development // Resolve #3652