platformio/platformio-core #3653: -std=gnu++2a causes cpp standard to be set to c++2 in c_cpp_properties.json

maciekr1234, 2020-09-05:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.0

### Description of problem

When I set standard to gnu++2a in GCC compiler pio doesn't match `a` which causes cpp extension to not set the correct standard.

[image]

I actually managed to track the issue and successfully fix the issue.

https://github.com/platformio/platformio-core/blob/96957203434f5e27f0f1bf0895371d380c3cf12a/platformio/ide/tpls/vscode/.vscode/c_cpp_properties.json.tpl#L71
```
% STD_RE = re.compile(r"\-std=[a-z\+]+([\da]+)")
```
https://github.com/platformio/platformio-core/blob/96957203434f5e27f0f1bf0895371d380c3cf12a/platformio/ide/tpls/vscode/.vscode/c_cpp_properties.json.tpl#L118
```
"cppStandard": "c++{{ cxx_stds[-1].replace('2a','20') }}",
```
This seems to solve the issue in cpp_config and keeps the compiler happy.
[image]

valeros, 2020-09-09:
Hi @maciekr1234 ! Thanks for reporting. It should be fixed in the development branch. Please upgrade to the latest revision via running `pio upgrade --dev` command in the IDE terminal and try again. Thanks!

closed 2020-09-09 by valeros

maciekr1234, 2020-09-09:
@valeros works great. Thanks.

commit 2a30ad0f (2020-09-09): Allow in-progress language standards in IDE templates // Resolve #3653

Note: VS Code only supports finalized names