platformio/platformio-core #3657: Platformio generates wrong CMake files for CLion

Herobone, 2020-09-07:
When using the command `platformio init --ide clion` it generates the two CMake files needed for CLion, but when opening CLion and loading the CMake Project the following error occurs:

`<command-line>:0:1: error: macro names must be identifiers`

It's not printed once but multiple times.
And I know what causes the problem. The command generates the following lines in CMakeListsPrivat.txt:

````
if (CMAKE_BUILD_TYPE MATCHES "due")
    add_definitions(-D'PLATFORMIO=50000')
    add_definitions(-D'__SAM3X8E__')
    add_definitions(-D'ARDUINO_SAM_DUE')
    add_definitions(-D'ARDUINO=10805')
    add_definitions(-D'F_CPU=84000000L')
    add_definitions(-D'USBCON')
    add_definitions(-D'USB_VID=0x2341')
    add_definitions(-D'USB_PID=0x003E')
    add_definitions(-D'USB_PRODUCT=\"Arduino Due\"')
    add_definitions(-D'USB_MANUFACTURER=\"Arduino\"')
    add_definitions(-D'ARDUINO_ARCH_SAM')```

BUT according to my knowledge and also what I found googeling this issue the code should look the following
(without single quotes after the -D)
```
if (CMAKE_BUILD_TYPE MATCHES "due")
    add_definitions(-DPLATFORMIO=50000)
    add_definitions(-D__SAM3X8E__)
    add_definitions(-DARDUINO_SAM_DUE)
    add_definitions(-DARDUINO=10805)
    add_definitions(-DF_CPU=84000000L)
    add_definitions(-DUSBCON)
    add_definitions(-DUSB_VID=0x2341)
    add_definitions(-DUSB_PID=0x003E)
    add_definitions(-DUSB_PRODUCT=\"Arduino Due\")
    add_definitions(-DUSB_MANUFACTURER=\"Arduino\")
    add_definitions(-DARDUINO_ARCH_SAM)
```
and in fact, removing these fixes the errors.

This is not only happening with the Due but as far as I tested also for Uno, Mega and Nano

My systems:

Windows 10 Professional Build 2004 (Slow insider channel)
and
Windows 10 Professional Build 1909

Visual Studio and Build Tools:
MSVS 2019 and 2017 (Community Edition)

CMake:
3.18.0 and 3.17.3

C and C++ Compiler:
LLVM clang-cl

Herobone, 2020-09-07:
Might be caused by https://github.com/platformio/platformio-core/blob/develop/platformio/ide/tpls/clion/CMakeListsPrivate.txt.tpl#L78

valeros, 2020-09-09:
Hi @Herobone ! Thanks for reporting, but it seems I cannot reproduce the issue on Windows with the latest CMake `3.18.2` and CLion `2020.2.1`. Could you please provide more info? What is your CLion and PlatformIO Core versions? Have you read our [doc page](https://docs.platformio.org/en/latest/integration/ide/clion.html) about using PlatformIO with CLion?

closed 2020-09-13 by ivankravets