platformio/platformio-core #3659: Error with compiling for ESP32

kugelkopf123, 2020-09-08:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
OSX 10.13.6 
**PlatformIO Version** (`platformio --version`):
PlatformIO on VSCode, version 5.0.0 
### Description of problem
I tried to compile the recent tasmota Dev Branch for the ESP32Dev Board. 
It stops with following Error:
*** [tasmota/StackThunk_light.cpp] /Volumes/Macintosh SD/Benutzer/myUsername/Downloads/Github/Tasmota/.pioenvs/tasmota32-DE/.sconsign37.dblite: No such file or directory

Ive tried to reinstall the whole PlatformIO/VSCode stuff. Deleted the .platformio Folder end tried it again. 
Nothing from that Works. 
I also tried the 2.0.0 Framework from Espressif32

#### Steps to Reproduce

1. Install VS Code + Platformio v5
2. Download Tasmota latestDev 
3. Try to Build tasmota32 Enviroment 

### Actual Results
```

> Executing task: platformio run --environment tasmota32-DE <

Processing tasmota32-DE (framework: arduino; platform: espressif32@2.0.0; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting tasmota.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 84 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32-to-ESP8266-compat> 0.0.2
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <HTTPUpdate> 1.3
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <BearSSL> 0.6
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
[... 638 lines cut ...]
      break;
    pos += 4;
  }
  ets_printf(">>>stack>>>\n");
  while (pos < stack_thunk_light_ptr) {
    ets_printf("%08x:  %08x %08x %08x %08x\n", (int32_t)pos, pos[0], pos[1], pos[2], pos[3]);
    pos += 4;
  }
  ets_printf("<<<stack<<<\n");
}

/* Called when the stack overflow is detected by a thunk.  Main memory is corrupted at this point.  Do not return. */
void stack_thunk_light_fatal_overflow()
{
    ets_printf("FATAL ERROR: BSSL stack overflow\n");
    abort();
}

};

```

### Additional info

Related issue from Tasmota issues: https://github.com/arendst/Tasmota/issues/9254

kugelkopf123, 2020-09-08:
Just tested it, its not related to ESP32 its the same with esp8266. Pls change the Topic.

ivankravets, 2020-09-08:
Please remove `/Volumes/Macintosh SD/Benutzer/myUsername/Downloads/Github/Tasmota/.pioenvs` folder and start building again. Does it work now?

kugelkopf123, 2020-09-08:
No. Nothing changed. Its the same. 
I also removed the .cache folder and tried it with "clean"  Its still the same.

ivankravets, 2020-09-09:
Do you have multiple PlatformIO Cores in a system? https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system

Do you use PlatformIO IDE and its built-in PIO Core or you installed it multiple times via pip/brew?

kugelkopf123, 2020-09-09:
> Do you have multiple PlatformIO Cores in a system? https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system
> 
> Do you use PlatformIO IDE and its built-in PIO Core or you installed it multiple times via pip/brew?

Thanks for your reply. 
I already checked this doc. No i havent.
I also tried the two commands to uninstall via pip/pip3/brew. But it says its not found on system. 
eg: 
`WARNING: Skipping platformio as it is not installed.`
or:
```
brew uninstall platformio
Error: No such keg: /usr/local/Cellar/platformio
```

ivankravets, 2020-09-09:
We are working on PlatformIO Core 5.0.1. It would be great to fix this issue. Could you mail me personally via me@ikravets.com? I'll provide further info how we could debug this issue. Thanks!

kugelkopf123, 2020-09-09:
@ivankravets Mail is on the way.

ivankravets, 2020-09-09:
Hi @arendst , @Jason2866,

Could you help here? Something going strange with a build process. It totally removes LIB and BUILD folders. I see also that whole project structure changes in runtime to legacy PIO Core 3.0 structure. New library folders triggers PlatformIO IDE to rebuild C/C++ IntelliSense index which leads to a fatal error. Someone removes internal PlatformIO folders in runtime.

ivankravets, 2020-09-09:
I forgot, it works from CLI but does not work from PlatformIO IDE for VSCode.

Jason2866, 2020-09-09:
@ivankravets  i am lost in this issue. I cant reproduce. Tasmota compiles without a error on Windows and Linux. Since i have no MAC i cant test on this platform. 
@s-hadinger (our zigbee developer) uses a MAC an do not have an issue too.

@kugelkopf123  `BearSSL`  is NOT supported in ESP32 . There is no port done.
Can you mail me your settings file (my_user_config.h and user_config_override.h and ALL platformio files).  obermeier.johann(at)gmail.com 
Maybe your Tasmota setup is `killing` platformio somehow.

kugelkopf123, 2020-09-09:
@Jason2866 
Ok done! But as i mentioned in the Mail, I just tried it once again with a fresh download in another folder without changing anything in any of the config files and its the same Problem.
I have it on both Macs. On my Macbook and my iMac.

Jason2866, 2020-09-09:
Used Github Actions CI to do a build on Mac OS. Builded Tasmota32-DE. No error
See https://github.com/Jason2866/Tasmota/runs/1093098093?check_suite_focus=true
Click on Tasmota32-DE version to see all LOGs (Details)

ivankravets, 2020-09-09:
It works from CLI but does not work from IDE. Do you remove “lib” or “.pioenvs” folders in extra scripts?

Jason2866, 2020-09-09:
Mhh, Just asked Stephan he uses VSC / Platformio and MacOs 10.14.6 Mojave. 
We have in lib folder our libraries (local). Many are modified. So no download from
Platformio registry or other online sources. Removing lib folder content is destroying Tasmota.
Removing .pioenvs is no problem. Rebuilded. @Staars (writing sensor driver for Tasmota) has a Mac too. His setup VSC / Platformio.  No idea what makes it fail

kugelkopf123, 2020-09-10:
Is there something more that i can check?

Jason2866, 2020-09-10:
okay, more tests...
Fresh „git clones“ on Catalina and Big Sur-beta are failing. For whatever reason the Tasmota32-folder is deleted by something. 
I did not find anything useful in the logs.
Compiling with MacOs 10.14.6 Mojave does work

kugelkopf123, 2020-09-10:
> okay, more tests...
> Fresh „git clones“ on Catalina and Big Sur-beta are failing. For whatever reason the Tasmota32-folder is deleted by something.
> not anything useful in the logs.
> Compiling with MacOs 10.14.6 Mojave does work
> @ivankravets There is a strange bug :-(

Ok. Good to hear that Iam not the only one. My Macbook is on 10.13.6HighSierra and my iMac on Mojave.
But i dont think that this have something to do with it. Dont know :-/.

ivankravets, 2020-09-10:
I could reproduce this issue. The problem is linked with invalid `platformio.ini`.

```
build_dir = .pioenvs
workspace_dir = .pioenvs
```

`build_dir` is a child of  `workspace_dir`, where you forced everything to 1 folder. PlatformIO checks for project modification and removed `build_dir` on changes. As result, it REMOVES EVERYTHING, including installed dependencies. They also are part of `workspace_dir`. I mean `lib_deps`. This causes PlatformIO IDE to rebuild IntelliSense Index every time because `lib_deps` modifies every time.

P.S: I removed `workspace_dir = .pioenvs` and now everything works. The question, why do you override the default build and workspace folders? Do we have a problem with PlatformIO Core?

kugelkopf123, 2020-09-10:
I can confirm, that works! Thanks! @ivankravets

Jason2866, 2020-09-10:
@ivankravets  Thx, for your help.  Yes it was a problem in the past. So it is a "left over". Will remove!!

Jason2866, 2020-09-10:
@kugelkopf123  Please close. Thx.

ivankravets, 2020-09-10:
This issue presents for any OS, not only macOS. Great that we found a problem.

Thanks, @kugelkopf123, for the remote session that helped to debug this issue and, @Jason2866, for the fix.

closed 2020-09-10 by ivankravets