platformio/platformio-core #3662: Libraries: Could not install library

WimBedet, 2020-09-09:
PIO Core Call Error: "Library Storage: /home/wim/Documents/PlatformIO/Projects/Sep5WeMosD1r1/.pio/libdeps/d1_mini\nLibrary Manager: Installing ottowinter/ESPAsyncWebServer-esphome @ ^1.2.7\nDownloading\nDownloading\nUnpacking\nUnpacking\nLibrary Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!\nLibrary Manager: Installing dependencies...\nLibrary Manager: Installing ESPAsyncTCP-esphome\nDownloading\nDownloading\nUnpacking\nUnpacking\nLibrary Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!\nLibrary Manager: Installing AsyncTCP-esphome\nDownloading\nDownloading\nUnpacking\nUnpacking\nLibrary Manager: AsyncTCP-esphome @ 1.1.1 has been installed!\nLibrary Manager: Installing Hash\n\n\nError: Could not find the package with 'Hash' requirements for your system 'linux_x86_64'"

luebbe, 2020-11-06:
Is this really fixed? Building with gh-workflows still throws the same error with PIO 5.0.2. Just tried again a few days ago.
See: https://github.com/luebbe/homie-node-collection/runs/1353074911?check_suite_focus=true

ivankravets, 2020-11-06:
@luebbe I can't reproduce it with your project

[image]

How to reproduce this issue?

luebbe, 2020-11-06:
@ivankravets It builds fine locally. I wanted to direct your attention to the output of the git action of this project. Maybe I made a mistake there, but exactly this error still shows up. I thought the link above would place you in the failed build step.

The same is true for homie itself. The last build on circleci failed with this error too:
https://app.circleci.com/pipelines/github/homieiot/homie-esp8266/33/workflows/8e1c3268-98da-4701-81db-4b86264691d5/jobs/1440

ivankravets, 2020-11-09:
It seems I know where the issue is hidden. Could you try to manually install the dev-platform with the Arduino framework?

```
pio platform install espressif8266 --with-all-packages
```

euphi, 2020-11-10:
I also had this problem and `--with-all-packages` seems to solve the problem (see https://travis-ci.org/github/euphi/ESP_Homie_WS2812FX/builds/742575326 - still red, but due to binary size issues.)

Obviously the error only occurs if this is a fresh install. Using a cache (for example in https://travis-ci.org/github/euphi/Homie_BareMinimum) avoids the issue for now. 

I guess more problems with built-in libraries will occur with the next update of espressif8266 platform update.

ivankravets, 2020-11-10:
It looks like a bug. We will fix it. Sorry.

luebbe, 2020-11-10:
Thank you @euphi. I received the notification from @ivankravets yesterday, but forgot to check. Sorry.

ivankravets, 2020-11-12:
Thanks for the report! PlatformIO Core 5.0.3 is going to be released today.

closed 2020-11-12 by ivankravets

mosagepa, 2021-07-09:
Any advances on this?
The compile and/our run executions "complain" about some libraries already been installed!...
I guess PlatformIO and all things related has grown to the point of overcomplicated mechanisms and rigmaroles... I miss the good'ol days of command line C compilations.

gknops, 2022-05-29:
macOS, this behavior is back. When I add Hash to lib_deps (just to see if that fixes it), I see
```

   Library Manager: Installing Hash
   Warning! Could not find the package with 'Hash' requirements for your system 'darwin_x86_64'

```

I assume the warning makes the build not fail, but doesn't mark the library as clean so it keeps trying to re-install it every time.

`pio platform install espressif32 --with-all-packages` fixed it for me.

commit 1f284e85 (2020-09-09): Fixed an issue when the package manager tries to install a built-in library from the registry // Resolve #3662
commit 4a12d195 (2020-11-12): Fixed an issue when the package manager tries to install a built-in library from the registry // Resolve #3662