platformio/platformio-core #3664: On automatic dependency installation fails with PermissionError: [WinError 5] Access is denied:

neilh10, 2020-09-10:
What kind of issue is this?

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 Pro latest

**PlatformIO Version** (`platformio --version`): Core 5.0.0·Home 3.3.0

### Description of problem
Whenever I install a dependency from GitHub repo (or full git url) the installation fails with:

PermissionError: [WinError 5] Access is denied: 'C:\\Users\\neilh77a\\.platformio\\.cache\\tmp\\pkg-installing-3k8x5w3o\\.git\\objects\\pack\\pack-c908a602e8bfab95b6b90a873267e2afcd16604b.idx':
  File "C:\Users\neilh77a\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

#### Steps to Reproduce

1. delete  .pio/libdeps
2  .press build button
3. after every failure, it does actually move the directory to the correct place,
4.  press build button again to have it try again

### Actual Results
Library Manager: Installing git+https://github.com/Seeed-Studio/Seeed_Arduino_LCD
git version 2.27.0.windows.1
Cloning into 'C:\Users\neilh77a\.platformio\.cache\tmp\pkg-installing-3k8x5w3o'...
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\neilh77a\\.platformio\\.cache\\tmp\\pkg-installing-3k8x5w3o\\.git\\objects\\pack\\pack-c908a602e8bfab95b6b90a873267e2afcd16604b.idx':
  File "C:\Users\neilh77a\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\neilh77a\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)

### Expected Results
That it installs all the listed dependencies

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```
[env:seeed_wio_terminal]
platform = atmelsam
board = seeed_wio_terminal
framework = arduino
lib_ldf_mode = deep+
build_flags = 
    ;-DSerial=SerialUSB
    ;-DSerial=STANDARD_SERIAL_OUTPUT
    ;-DSTANDARD_SERIAL_OUTPUT=Serial4

lib_deps= 
    ;https://github.com/Seeed-Studio/Seeed_Arduino_atWiFi
    ;https://github.com/Seeed-Studio/Seeed_Arduino_FreeRTOS
    ;https://github.com/Seeed-Studio/Seeed_Arduino_atUnified
    ;https://github.com/Seeed-Studio/esp-at-lib
    ;https://github.com/Seeed-Studio/Seeed_Arduino_mbedtls
    ;https://github.com/Seeed-Studio/Seeed_Arduino_atWiFiClientSecure
    https://github.com/Seeed-Studio/Seeed_Arduino_LCD
    https://github.com/Seeed-Studio/Seeed_Arduino_RTC
    https://github.com/Seeed-Studio/Seeed_Arduino_FS
    https://github.com/Seeed-Studio/Seeed_Arduino_SFUD
    https://github.com/adafruit/Adafruit_ZeroTimer

**Source file to reproduce issue:**
```cpp
Insert here...
```
#include <Seeed_FS.h>
void setup(){}
void loop(){}

### Additional info
I've been using PIO for two years, and its been fantastic. Thankyou.
This may have happened after the upgrade to 5.0.0 - at least it wasn't as bad as this last week.
I've just started trying a new platform wio, and in the process upgraded the Platform to SAMD 4.5.1 but I don't think it is related to that
I've looked around for should I completely reinstall PIO
When opening another project, it says PlatformIO: Can not find working Python 2.7 or 3.5+ interpreter. Please install the latest Python 3 and restart VSCode . Trying to track down that bug and rebooting.

ivankravets, 2020-09-12:
Please re-test with `pio upgrade --dev`. Thanks for the report!

closed 2020-09-12 by ivankravets

neilh10, 2020-09-12:
Worked. Many thanks. Fantastic PlugIn. Don't know how you do it, but its magic.!!!

> pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.0.2a2
Release notes: https://docs.platformio.org/en/latest/history.html
Warning! Please restart IDE to affect PIO Home changes

ivankravets, 2020-09-12:
Happy coding with PlatformIO 😊

commit 687c339f (2020-09-12): Fixed a "PermissionError: [WinError 5]" on Windows when external repository is used with `lib_deps` option // Resolve #3664