platformio/platformio-core #3666: Could not load platform updates: KeyError: 'versions'

VIXIVIXIV, 2020-09-11:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n    cli()  # pylint: disable=no-value-for-parameter\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n    rv = self.invoke(ctx)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n    return super(PlatformioCLI, self).invoke(ctx)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 393, in platform_update\r\n    and not PlatformFactory.new(pkg).are_outdated_packages()\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\platform\\_packages.py\", line 134, in are_outdated_packages\r\n    pkg, self.get_package_spec(pkg.metadata.name)\r\n  File \"C:\\Users\\Ben\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_update.py\", line 48, in outdated\r\n    self.pick_best_registry_version(reg_pkg[\"versions\"]) or {}\r\nKeyError: 'versions'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2020-09-11:
Please open PlatformIO CLI and type `pio upgrade --dev`. Does it work now?

> NTPClient=lib/NTPClient

@Majubs , this is wrong declaration. See valid formats https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html

closed 2020-09-11 by ivankravets

Majubs, 2020-09-11:
It seems to be building fine now 😃 Thanks!

Altough I'm still getting the warning about the libs. I made some modifications to the NTPClient Arduino library, that's why I placed in `/lib` folder and added the `NTPClient=lib/NTPClient` to platform.ini. Same with ESP8266Ping. It always worked fine this way.

I'm not sure what I need to do to get rid of the warning. Do I need to install them (as explained in the linked docs) even if I'm not using the version from Arduino? Or is it something else?

ivankravets, 2020-09-12:
Just remove 
```
  NTPClient=lib/NTPClient
  ESP8266Ping=lib/ESP8266Ping
```
You re-install libs twice. They are already in your lib folder.

Majubs, 2020-09-14:
Oh, ok. Dropped the lines in `platform.ini` and it seems to be working fine. I thought that it would use the Arduino version of the libraries without it, but I guess platformio prioritizes the `/lib` folder when searching for libs 😁

VIXIVIXIV, 2020-09-20:
@ivankravets Sorry for not getting back to you earlier, but after upgrading to the dev build of PlatformIO, the issue with updating platforms is resolved; I was able to upgrade framework-arduino-lpc176x from 0.2.4 to 0.2.5

commit 7bc170a5 (2020-09-11): Fixed an issue with "KeyError: 'versions'" when dependency does not exist in the registry // Resolve #3666