platformio/platformio-core #3667: Failed to build cryptography on RPi

mikesowter, 2020-09-13:
On a new build RPi, "platformio remote agent start" fails with the "Failed to build cryptography" message

I have run the "sudo apt install python3-dev libffi-dev" fix for FFI files

is there a new fix for this problem? I couldn't find it by searching this repo.

see below error message in detail:

Building wheel for cryptography (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.platformio/penv/bin/python /home/pi/.platformio/penv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmp96760vn3
       cwd: /tmp/pip-install-l4v76t55/cryptography
  Complete output (137 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/cryptography
  copying src/cryptography/__init__.py -> build/lib.linux-armv7l-3.7/cryptography
  copying src/cryptography/utils.py -> build/lib.linux-armv7l-3.7/cryptography
  copying src/cryptography/fernet.py -> build/lib.linux-armv7l-3.7/cryptography
  copying src/cryptography/__about__.py -> build/lib.linux-armv7l-3.7/cryptography
  copying src/cryptography/exceptions.py -> build/lib.linux-armv7l-3.7/cryptography
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/_oid.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/_der.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat
  creating build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/certificate_transparency.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/base.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/ocsp.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/oid.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/extensions.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/general_name.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  copying src/cryptography/x509/name.py -> build/lib.linux-armv7l-3.7/cryptography/x509
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/interfaces.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/poly1305.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/primitives
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings
  copying src/cryptography/hazmat/bindings/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ec.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/hashes.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ciphers.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/x25519.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/aead.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/utils.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/dsa.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ocsp.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/rsa.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/backend.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/x509.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ed448.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/encode_asn1.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/cmac.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ed25519.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/decode_asn1.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/x448.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
[... 115 lines cut ...]
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/commands/remote/command.py", line 48, in cli
    inject_contrib_pysite(verify_openssl=True)
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 90, in inject_contrib_pysite
    build_contrib_pysite_deps(get_core_package_dir("contrib-pysite"))
  File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 115, in build_contrib_pysite_deps
    dep,
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.platformio/penv/bin/python', '-m', 'pip', 'install', '--no-compile', '--no-binary', ':all:', '-t', '/home/pi/.platformio/packages/contrib-pysite', 'autobahn == 20.4.3']' returned non-zero exit status 1.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2020-09-13:
> #include <openssl/opensslv.h>

https://stackoverflow.com/questions/3016956/how-do-i-install-the-openssl-libraries-on-ubuntu

closed 2020-09-13 by ivankravets

mikesowter, 2020-09-14:
I think the build line at fault is:
_File "/home/pi/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 88, in inject_contrib_pysite from OpenSSL import SSL_
I do not end up with an OpenSSL folder in contrib-pysite
I have the latest openssl installed by pip

I do not understand where to put your suggested #include <openssl/opensslv.h> because I can't see any C files in the build process.  I still am inexperienced at python, although finding it increasing vital to debugging this stuff

mikesowter, 2020-09-14:
ok, I can see in: 
  ___build/temp.linux-armv7l-3.7/_openssl.c:575:10: fatal error: openssl/opensslv.h: No such file or directory
   #include <openssl/opensslv.h>___

that it didn't compile, but where is the temp.linux-armv7l-3.7/_openssl.c sourced from? I am not familiar with the build process!

mikesowter, 2020-09-14:
Ok, "apt-get install libssl-dev" got me a lot further, but now I get:
**_ImportError: cannot import name '_get_backend' from 'cryptography.hazmat.backends' (/home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/backends/__init__.py)_**

ivankravets, 2020-09-14:
Could you try now to run `pio remote agent start` command?

mikesowter, 2020-09-15:
Thanks Ivan, I just assumed it wouldn't run, but it did! I am a bit nervous about using all these dev libraries: _python3-dev libffi-dev libssl-dev_, on other platforms they distinguish between full releases, candidates, alpha and beta releases... where do you think these -dev libraries fit in that sort of scheme?

ivankravets, 2020-09-15:
It looks like our default pre-built packages are not compatible with your ARM-based device. In this case, PlatformIO tries to build packages from sources. This is why it needs `python3-dev libffi-dev libssl-dev`