platformio/platformio-core #3668: LDF regression between 4.3.4 and 5.0.1

mairas, 2020-09-14:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

MacOS 10.15.6

**PlatformIO Version** (`platformio --version`):

4.3.4 and 5.0.1

### Description of problem

I've noticed that the new library dependency finder logic doesn't properly pick up all dependencies in a simple project using SensESP (https://github.com/SignalK/SensESP).

My test project is this one: https://github.com/mairas/sk-engine-temp

Specifically, `sk-engine-temp` depends on SensESP (via a GitHub link, not just a library name - we've done a horrible job at making timely releases...). SensESP depends on multiple Adafruit flavor libraries, for example `Adafruit BME280 Library`. That library (and its siblings) has a `library.properties` which defines a dependency to `Adafruit Unified Sensor`. That dependency does get picked up by PIO 4.3.4 but not by PIO 5.0.1.

#### Steps to Reproduce

1. In the project directory:
```
rm -rf .pio
~/.platformio/penv/bin/pip install platformio==4.3.4
pio run
```
succeeds.
2.
```
rm -rf .pio
~/.platformio/penv/bin/pip install platformio==5.0.1
pio run
```
fails.
3. There is no step 3!

### Actual Results

```
Compiling .pio/build/d1_mini/lib3e1/Adafruit BMP280 Library/Adafruit_BMP280.cpp.o
In file included from .pio/libdeps/d1_mini/Adafruit BMP280 Library/Adafruit_BMP280.cpp:22:0:
.pio/libdeps/d1_mini/Adafruit BMP280 Library/Adafruit_BMP280.h:24:29: fatal error: Adafruit_Sensor.h: No such file or directory

*************************************************************************
* Looking for Adafruit_Sensor.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_Sensor.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_Sensor.h
*
*************************************************************************

 #include "Adafruit_Sensor.h"
                             ^
compilation terminated.
```

### Expected Results

```
======================== [SUCCESS] Took 220.77 seconds ========================
```
(could be a bit faster too)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

As in https://github.com/mairas/sk-engine-temp

**Source file to reproduce issue:**

As in https://github.com/mairas/sk-engine-temp

### Additional info

PlatformIO is awesome, nevertheless!

ivankravets, 2020-09-16:
It was a bug in PlatformIO 4.0. We already discussed it. PlatformIO uses `chain+` mode for Arduino libraries now.

See https://community.platformio.org/t/library-dependency-problem-related-to-core-5-0/15726

mairas, 2020-09-16:
I don't think this is the same issue: this was something that worked in PIO 4.3.4 but fails in 5.0.1. Also, I checked both Adafruit_BME280.h and Adafruit_BMP280.h, both included from respective files in SensESP, and the Adafruit headers do include Adafruit_Sensor.h which is the main include file for "Adafruit Unified Sensor". I think LDF _should_ pick that up.

ivankravets, 2020-09-16:
@valeros, please help

mairas, 2020-09-16:
I'm copy-pasting from https://community.platformio.org/t/library-dependency-problem-related-to-core-5-0/15726/5:

If I don't define `lib_ldf_mode`, the default mode seems to be `chain`:

```
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

That doesn't work and produces results described in the issue.

However, `deep` works. Then again, `chain+` doesn't work but this time it's missing `ESP8266WiFi.h` (which _is_ present in the dependency graph). Finally, `deep+` fails due to the same problem as `chain+`.

The plus finder modes probably fail due to conditional importing such as the one in `ESPAsyncWiFiManager.h`:

```
#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
```

However, my target for the build is `d1_mini` which definitely has `ESP8266` defined.

mairas, 2020-09-17:
I've updated SensESP dependencies to workaround this bug. For testing purposes, I've made `sk-engine-temp` to depend on SensESP commit 54ceaced50511aa09c148563c080508876c88b38 for your convenience.

Rotzbua, 2020-12-05:
I ran into a similiar problem.

My library uses just the `library.properties` where the dependencies are defined.

The dependencies are just the buildin libraries wire and spi. So it is defined as `depends=Wire,SPI`.
With pio `5.03` it just resolves `SPI` as dependency, but with latest pio `4` it resolves both dependencies and works as expected.

I think that with the new dependency management something is broken with the arduino style compatibility.

ivankravets, 2020-12-22:
@Rotzbua , please add this code to your project.

```c
#include <SPI.h>
#include <Wire.h>
```

Rotzbua, 2020-12-22:
@ivankravets It is included in the library header. Adding it additionally to the main (example file in my case) should not be required.

Rotzbua, 2020-12-22:
I think I observed problem:

1. Library structure
The library contains two drivers. One for i2c and one for spi: `DriverI2C.h` and `DriverSPI.h`.
The  `DriverI2C.h` contains `#include <Wire.h>` and `DriverSPI.h` contains `#include <SPI.h>`.

2. Test cases
If the example includes just one of the `DriverXXX.h` then I get an compilation error.
If both are included it compiles fine.

3. Problem
So the LDF just looks for the used libraries in the **included** source files , but then compiles **all** source files. So it fails due non resolved libraries for **all** source files.

ivankravets, 2021-03-15:
Is this issue fixed in the latest PlatformIO Core?