platformio/platformio-core #3669: Native environment linker problems after upgrading to 5.0.1

wrong-kendall, 2020-09-15:
### Configuration

**Operating system**:
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=19.1
DISTRIB_CODENAME=tessa
DISTRIB_DESCRIPTION="Linux Mint 19.1 Tessa"

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 5.0.1

### Description of problem
When trying to run a test file that previously worked on 4.3.4, the linker fails to link. I specifically noticed this while using ArduinoFake but suspect it happens in other cases too (but haven't verified). I used the test case below and 2 separate virtualenvs to verify that the test runs on 4.3.4 but not on 5.0.1

#### Steps to Reproduce

1. Create header that conditionally includes ArduinoFake.h
2. Make a class with a simple method that calls one of the Arduino builtin functions
3. Create a test that includes ArduinoFake.h, instantiates the class created in step 2 and tries to call that method.

### Actual Results
```
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
.pio/build/native/lib1de/libFoo.a(Foo.o): In function `Foo::DoStuff()':
Foo.cpp:(.text+0x17): undefined reference to `pinMode'
collect2: error: ld returned 1 exit status
*** [.pio/build/native/program] Error 1
================================================================ [FAILED] Took 2.18 seconds ================================================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       m4             IGNORED
*       native         FAILED    00:00:02.184
=========================================================== 1 failed, 0 succeeded in 00:00:02.184 ===========================================================
```
### Expected Results
```
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
---------------------------------------------------------------------------------------------------------
Building...
CorePackageManager: Installing tool-scons @ ~3.30102.0
tool-scons @ 3.30102.0 has been successfully installed!
Uninstalling tool-scons @ 4.40001.0: 	[OK]
Testing...
test/test_main.cpp:20:test_truth	[PASSED]

-----------------------
1 Tests 0 Failures 0 Ignored
OK
====================================== [PASSED] Took 3.26 seconds ======================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       m4             IGNORED
*       native         PASSED    00:00:03.265
====================================== 1 succeeded in 00:00:03.265 ======================================
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:m4]
platform = atmelsam
board = adafruit_feather_m4
framework = arduino

[env:native]
platform = native
build_flags = -std=gnu++11
lib_deps =
    ArduinoFake
lib_compat_mode = off
```

**Source file to reproduce issue:**
**Foo.h**
```cpp
#pragma once
#ifdef UNIT_TEST
#include "ArduinoFake.h"
#else
#include "Arduino.h"
#endif

class Foo {
public:
  Foo(){};
  void DoStuff();
};
```
**Foo.cpp**
```cpp
#include "Foo.h"

void Foo::DoStuff() {
    pinMode(1, HIGH);
    return;
}
```
**test_main.cpp**
```cpp
#include <ArduinoFake.h>
#include <unity.h>

using namespace fakeit;

#include "Foo.h"

void setUp(void) { ArduinoFakeReset(); }

void test_kaboom() {
  Foo f;
  f.DoStuff();
}

void test_truth() { TEST_ASSERT_TRUE(true); }
int main(int argc, char **argv) {
  UNITY_BEGIN();
  RUN_TEST(test_truth);
  return UNITY_END();
}
```
### Additional info
Here are the verbose runs for each.
[linker_verbose_failure.txt](https://github.com/platformio/platformio-core/files/5228666/linker_verbose_failure.txt)
[linker_verbose_success.txt](https://github.com/platformio/platformio-core/files/5228667/linker_verbose_success.txt)

ivankravets, 2020-09-17:
Please re-test with the latest PIO Core dev version via `pio upgrade --dev`.

closed 2020-09-17 by ivankravets

wrong-kendall, 2020-09-17:
Confirmed success. Thanks!
[11:12:22 09/17/20]: <~/development/src/linker_woes>$ pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
test/test_main.cpp:20:test_truth        [PASSED]

-----------------------
1 Tests 0 Failures 0 Ignored
OK
============================================================================ [PASSED] Took 0.33 seconds ============================================================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       m4             IGNORED
*       native         PASSED    00:00:00.331
============================================================================ 1 succeeded in 00:00:00.331 ============================================================================

commit 1e0ca8f7 (2020-09-17): Fixed an issue with GCC linker when "native" dev-platform is used in pair with library dependencies // Resolve #3669