platformio/platformio-core #3674: 5.0.1 build no longer uses start-group/end-group linker flags (on native platform)

nomis, 2020-09-16:
### Configuration

**Operating system**: Ubuntu 18.04

**PlatformIO Version** (`platformio --version`): 5.0.1

### Description of problem

Builds with the native platform no longer pass `start-group`/`end-group` around libraries specified as linker arguments, which is causing my builds to fail. If I manually add these back then the final link step works. Builds on other platforms always use these arguments. This worked on 4.3.4.

Regression from #3282.

#### Steps to Reproduce
1. git clone https://github.com/nomis/mcu-uuid-console
2. cd mcu-uuid-console/test
3. cd build/example-DigitalIO; pio lib install uuid-common uuid-log; cd -
4. platformio run -d build/example-DigitalIO -e native -v

(at commit c9d04baf7143f0eef5826f4c79c55841bff9a7d0)

### Actual Results
```
...
c++ -o .pio/build/native/program .pio/build/native/src/Arduino.o .pio/build/native/src/DigitalIO.o -L.pio/build/native .pio/build/native/lib8dc/libuuid-common.a .pio/build/native/libbdb/libuuid-log.a .pio/build/native/lib380/libuuid-console.a
.pio/build/native/libbdb/libuuid-log.a(log.o): In function `std::_Sp_counted_ptr_inplace<uuid::log::Message, std::allocator<uuid::log::Message>, (__gnu_cxx::_Lock_policy)2>::_M_get_deleter(std::type_info const&)':
log.cpp:(.text._ZNSt23_Sp_counted_ptr_inplaceIN4uuid3log7MessageESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE14_M_get_deleterERKSt9type_info[_ZNSt23_Sp_counted_ptr_inplaceIN4uuid3log7MessageESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE14_M_get_deleterERKSt9type_info]+0xb): undefined reference to `typeinfo name for std::_Sp_make_shared_tag'
.pio/build/native/libbdb/libuuid-log.a(log.o): In function `std::shared_ptr<uuid::log::Message> std::make_shared<uuid::log::Message, unsigned long, uuid::log::Level&, uuid::log::Facility&, __FlashStringHelper const* const&, char*>(unsigned long&&, uuid::log::Level&, uuid::log::Facility&, __FlashStringHelper const* const&, char*&&)':
log.cpp:(.text._ZSt11make_sharedIN4uuid3log7MessageEJmRNS1_5LevelERNS1_8FacilityERKPK19__FlashStringHelperPcEESt10shared_ptrIT_EDpOT0_[_ZSt11make_sharedIN4uuid3log7MessageEJmRNS1_5LevelERNS1_8FacilityERKPK19__FlashStringHelperPcEESt10shared_ptrIT_EDpOT0_]+0x152): undefined reference to `typeinfo for std::_Sp_make_shared_tag'
.pio/build/native/libbdb/libuuid-log.a(log.o):(.data.rel.ro._ZTISt23_Sp_counted_ptr_inplaceIN4uuid3log7MessageESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE[_ZTISt23_Sp_counted_ptr_inplaceIN4uuid3log7MessageESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE]+0x10): undefined reference to `typeinfo for std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>'
.pio/build/native/libbdb/libuuid-log.a(format_timestamp_ms.o): In function `uuid::log::format_timestamp_ms[abi:cxx11](unsigned long, unsigned int)':
format_timestamp_ms.cpp:(.text+0xb0): undefined reference to `snprintf_P(char*, unsigned long, char const*, ...)'
```

### Expected Results
```
...
c++ -o .pio/build/native/program .pio/build/native/src/Arduino.o .pio/build/native/src/DigitalIO.o -L.pio/build/native -Wl,--start-group .pio/build/native/lib8dc/libuuid-common.a .pio/build/native/libbdb/libuuid-log.a .pio/build/native/lib380/libuuid-console.a -Wl,--end-group
============================================================================================================================ [SUCCESS] Took 2.55 seconds ============================================================================================================================
```

nomis, 2020-09-16:
This commit is the cause:
```
commit 74e27a2edc94e9531eaf81bb7047199f6a7a9296
Author: Ivan Kravets <me@ikravets.com>
Date:   Sun Aug 16 20:26:59 2020 +0300

    Enable "cyclic reference" for GCC linker only for the embedded dev-platforms // Resolve #3570

diff --git a/HISTORY.rst b/HISTORY.rst
index 147c4b5f..7f86e40f 100644
--- a/HISTORY.rst
+++ b/HISTORY.rst
@@ -54,6 +54,7 @@ PlatformIO Core 4
 * Dump data intended for IDE extensions/plugins using a new `platformio project idedata <https://docs.platformio.org/page/core/userguide/project/cmd_idedata.html>`__ command
 * Do not generate ".travis.yml" for a new project, let the user have a choice
 * Automatically enable LDF dependency `chain+ mode (evaluates C/C++ Preprocessor conditional syntax) <https://docs.platformio.org/page/librarymanager/ldf.html#dependency-finder-mode>`__ for Arduino library when "library.property" has "depends" field (`issue #3607 <https://github.com/platformio/platformio-core/issues/3607>`_)
+* Enable "cyclic reference" for GCC linker only for the embedded dev-platforms (`issue #3570 <https://github.com/platformio/platformio-core/issues/3570>`_)
 * Updated PIO Unit Testing support for Mbed framework. Added compatibility with Mbed OS 6
 * Do not escape compiler arguments in VSCode template on Windows
 * Fixed an issue with PIO Unit Testing when running multiple environments (`issue #3523 <https://github.com/platformio/platformio-core/issues/3523>`_)
diff --git a/platformio/builder/tools/platformio.py b/platformio/builder/tools/platformio.py
index 560cbe37..5d8f8e8b 100644
--- a/platformio/builder/tools/platformio.py
+++ b/platformio/builder/tools/platformio.py
@@ -66,7 +66,11 @@ def BuildProgram(env):
         env.Prepend(LINKFLAGS=["-T", env.subst("$LDSCRIPT_PATH")])
 
     # enable "cyclic reference" for linker
-    if env.get("LIBS") and env.GetCompilerType() == "gcc":
+    if (
+        env.get("LIBS")
+        and env.GetCompilerType() == "gcc"
+        and env.PioPlatform().is_embedded()
+    ):
         env.Prepend(_LIBFLAGS="-Wl,--start-group ")
         env.Append(_LIBFLAGS=" -Wl,--end-group")
 
```

rhapsodyv, 2020-09-16:
Only Mac don't have support for `-Wl,--start-group/-Wl,--end-group`

nomis, 2020-09-16:
Is there an equivalent linker process for that - what happens if you try to run my native build on macOS?

I could have Travis CI do a build but it's not interactive so it would be tedious to try to get it working.

rhapsodyv, 2020-09-16:
Nop. 

Now I really don't know if Mac just don't have the ordering issue. I can test it.

ivankravets, 2020-09-17:
Sorry, we will revert back that commit. It's actually a bug. I don't know why removed these linker flags. It seems that we have an issue with the incorrect detecting of GCC. Need to check which version does not support this.

I mark this issue as duplicated and will make a fix. 

Duplicate of https://github.com/platformio/platformio-core/issues/3669

closed 2020-09-17 by ivankravets

ivankravets, 2020-09-17:
Please re-test with the latest PIO Core dev version via `pio upgrade --dev`.

tsandmann, 2021-03-23:
As `env.PioPlatform().is_embedded()` returns true only if the packages list from platform.json contains a package of type *uploader*, the issue still exists on macOS for platforms that don't specify an uploader tool.

base.py:
```python
def is_embedded(self):
    for opts in self.packages.values():
        if opts.get("type") == "uploader":
            return True
    return False
```

Why does `is_embedded()` rely on the existence of an uploader tool for an embedded platform?

ivankravets, 2021-03-24:
>  the issue still exists on macOS for platforms 

Could you point us to the affected dev-platforms?

tsandmann, 2021-03-24:
At least [rpi-pico](https://github.com/Wiz-IO/wizio-pico) and [apollo3](https://github.com/nigelb/platform-apollo3blue) had or have this issue. Maybe there are more or will be more in the future, because AFAIK the PIO documentation doesn't state that platform.json must contain a package of type *uploader* to configure the linker to use `-Wl,--start-group` / `-Wl,--end-group` parameters on macOS hosts.

ivankravets, 2021-03-24:
You need to file issues in their repositories. They can implement 

```
def is_embedded(self):
    return True
```

in https://github.com/Wiz-IO/wizio-pico/blob/main/platform.py#L13

tsandmann, 2021-03-24:
Thanks, that's good to know.

ivankravets, 2021-03-24:
@tsandmann I've just made a PR https://github.com/Wiz-IO/wizio-pico/pull/60