platformio/platformio-core #3681: PlatformIO Home does not work over SSH tunnel

ivankravets, 2020-09-22:
VSCode Remote development / LInux SSH.

septatrix, 2021-03-21:
Is known what parts exactly are the culprit?

sergerold, 2021-04-18:
any update on this? would be great to code using PIO on a remote device using VS Code

btw, i just came across PIO and absolutely love it!

nelsonov, 2021-11-06:
I have noticed that when PlatformIO creates folders on the remote, it uses the windows "\" as a directory separator rather than the Unix "/". So, for me it I already have a path `/home/username/Documents/PlatformIO/Projects` with several projects below it from developing locally on that remote.  However, when using ssh-remote from my Windows laptop and I try to create a new project, it escapes the backslash and creates the path `/home/lnelson/Documents/PlatformIO/Projects\\NewProject`.  This is a directory called `Projects\\NewProject`, parallel to `Projects`, not a directory `NewProject` under `Projects`. This probably also explains why I can navigate withing VSCodes remote file explore but can't traverse directories when looking for files in PlatformIO's "Add Existing" and similar commands.

When prompted by VSCode's ssh-remote for the system type, I do specify "Linux", so somewhere in the depth's of ssh-remote, it knows I connected to a unix system.

nelsonov, 2021-11-06:
Also, when I fix the path on the remote system so that I have `...Projects/NewProject` instead of `...Projects\\NewProject`, I'm still unable to open it.  When I go to "Open Existing" to open the path to `NewProject`, it opens the directory but the path shown in the top of the window is `Projects / NewProject` (with the spaces) and it's unable to find `platformio.ini` even thought it's right there in front of my eyes.

nelsonov, 2021-11-07:
When I run the Linux-native version of VSCode inside WSL on the same machine, then everything works correctly.  It looks to me like the Windows version of PlatformIO is having trouble composing unix-style paths when using remote-ssh.

plinnie, 2022-03-17:
I have the same problem using VSCode on Windows  10 an connection using Remote-SSH to my Linux (Ubuntu 20.04) system, to which the board is attached. It would be great if you managed to get this working.

nische, 2022-06-28:
Hi, 
Do you have some new info about it or a workaround? 

BR, NiSche

elestedt, 2022-09-26:
I see the same with VSCode on Windwos, connecting to a WSL2 Ubuntu 22.04.1
At a glance this seems related to #3963 in where the webinterface run on a linux machine, but accessed from a windows machine detects the operating system of the web browser and not where the application is actually running.

delhub, 2022-10-15:
has anyone come up with a solution or a workaround?

cbundy, 2022-10-23:
> has anyone come up with a solution or a workaround?

I have a sample PR to fix this, which has the frontend determine the path style to use based on the PIO backend, rather than the OS the browser is running in.

I've not contributed here, I'm not sure if/when we could expect to see a maintainer assist with it.

https://github.com/platformio/platformio-home/pull/5018

CDejoEcobee, 2024-03-14:
Has there been any update on this?

Possibly adding a setting to the extension that'll force unix style directories?

cbundy, 2024-03-15:
I made a PR but I've closed it as the owners want to fix this in an upcoming major release

skullcrushercmd, 2024-04-07:
> I made a PR but I've closed it as the owners want to fix this in an upcoming major release

could you share the change so i can fix temporarily?

CDejoEcobee, 2024-04-17:
@skullcrushercmd you can find the pull request with the fix at https://github.com/platformio/platformio-home/pull/5018/commits

aviralpatel, 2025-02-28:
### Here's how you can fix it (Debian)

- Open VScode remote server
- Type `pio home --no-open`
- If above command is not found add `~/.platformio/penv/bin` to path variable
- When above command is running, look for the port it is running on
- You can forward this port by pressing **cmd+shift+p** and then typing **forward a port**
- Or to make this change permanent go to ~/.ssh/config and add `LocalForward 127.0.0.1:PORT 127.0.0.1:PORT` parameter
- Now when the port is forwarded open the link in your browser to create a new project
- Once the project is created use **open a folder** button to open it

LucaUrbinati44, 2025-08-01:
Thanks to @aviralpatel, I solved the problem just by inserting this line `export PATH=~/.platformio/penv/bin:$PATH` into my ~/.bashrc file. Thank you