platformio/platformio-core #3688: Private library dependencies not honored

programminghoch10, 2020-09-27:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version**: 5.0.1

### Description of problem

External Library dependencies specified in `library.json` of private librarys are ignored by the library dependency finder.

#### Steps to Reproduce

1. Create a new project
1. Add folder named `Foo` in `/lib`
1. Inside it add `Foo.h` and `library.json`
    `Foo.h`:
    ```cpp
    #include "ArduinoJson.h"
    ```
    `library.json`:
    ```cpp
    {
        "dependencies": {
            "bblanchon/ArduinoJson" : "*"
        }
    }
    ```
   Foo now depends on ArduinoJson and also includes it.
1. Put `#include <Foo.h>` inside `src/main.cpp`
1. Compile it

### Actual Results
```
Dependency Graph
|-- <Foo>
```

### Expected Results
```
Dependency Graph
|-- <Foo>
|   |-- <ArduinoJson> 6.16.1
```

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

### Additional info
When ArduinoJson is installed already, it does find the dependencies correctly.

melkentum, 2020-10-06:
I am working on a project where we are currently trying to move our code from the `src` directory into separate private libraries. We encountered exactly the same problem, which is truely a bummer. 

If I run `pio run --verbose` I get warnings of the following kind:
```
Warning: Ignored `Foo` dependency for `Bar` library
```

This warning is emitted by `LibBuilderBase::process_dependencies` which iterates over all dependencies specified in `library.json`: 
```python
def process_dependencies(self):
    if not self.dependencies:
        return
    for item in self.dependencies:
        found = False
        for lb in self.env.GetLibBuilders():
            if item["name"] != lb.name:
                continue
            found = True
            if lb not in self.depbuilders:
                self.depend_recursive(lb)
            break
    
            if not found and self.verbose:
                sys.stderr.write(
                    "Warning: Ignored `%s` dependency for `%s` "
                    "library\n" % (item["name"], self.name)
                )
```

As far as I understand this, within the loop it is checked whether for a given specified dependency a library in the list provided by `self.env.GetLibBuilders()` exists. The issue here I think is that this list only contains libraries that are either already installed or otherwise locally provided via e.g. the `lib` directory. In case of libraries which are provided by other means, e.g. via PlatformIO's library registry or via Git URLs, the code above fails and prints the aforementioned warning.

vortigont, 2020-10-20:
Look like a duplicate for #2573

closed 2020-10-27 by valeros