platformio/platformio-core #3689: [Feature] Remote Development without PIO account

septatrix, 2020-09-27:
What kind of issue is this?

- [ ] **Question**.
- [ ] **PlatformIO IDE**.
- [ ] **Development Platform or Board**.
- [X] **Feature Request**.
- [ ] **PlatformIO Core**.

------------------------------------------------------------------

### Description of problem

At my university we would like to use PIO for some courses. For some tasks we provide the students with remote development boards which they can currently access over a virtual machine with a programmer and webcam attached. However such vms are relatively slow and have a high latency as the IDE and tooling also run on the vm.

The remote development capabilities would be a perfect replacement for this however we cannot require our students to create PIO accounts. Therefore it would be very nice if one was instead able to just provide the IP of the machine running the remote agent instead of needing an account setup on both machines. Optionally it would be nice to be able to protect the remote agent with a password however for our use case this is all set up using firewall rules.

Another alternative solution I stumbled upon in the forum would be to enable people to set up their own broker (i.e. the PIO Cloud in the [technology overview](https://docs.platformio.org/en/latest/plus/pio-remote.html#technology)) which is where the agents advertise themselves to and what the clients query to get remote devices. This would also be beneficial for larger setups where the IP is unknown but the local network is disconnected from the internet.

### Additional info

Related discussions in the forum: [Pio remote without login?](https://community.platformio.org/t/pio-remote-without-login/15258) and [In-house remote server](https://community.platformio.org/t/in-house-remote-server/14362)

Pinging @ivankravets as he apparently knows best how this works and said that they would look into this.

septatrix, 2020-10-07:
Any thoughts on this?

ivankravets, 2020-10-14:
It's impossible, we need a broker, see how does it work https://docs.platformio.org/en/latest/plus/pio-remote.html#technology

septatrix, 2020-10-16:
Yes, **currently** this is not possible. That is exactly what issues are made for.

While I understand that a broker is necessary for things to work this does not mean that such a broker could not be deployed locally or be used without an PIO account. Due to local laws we (understandably) cannot require students to create a third party account for attend a course.

ivankravets, 2020-10-16:
Thanks, I reopened this issue.

lit1s, 2020-11-14:
Dear sirs. First of all, many thanks for PlatformIO. It's Huge!!!!! :-D Looing to improve my experience, I've recently look for that. My setup: my main (and noisy) computer, is in the "server" room, with about 12 meter of 4 network (STP) cables, to extend HDMI and two USB's. One USB, for the keyb. and mouse, with a hub, (and works!) and the other usb, to program the Arduinos, Teensy's, Seeed's, etc... but it works about 1 of every 2 or 3 attempts. So I was looking to "remote" upload sketchs, on a small (and quiet) laptop on my desk... And had some success with "usbip" (I use Debian around)... but was looking for someting more natural... Waiting for an easy something like "remote dev. and sketch upload for Local LAN offline broker-less"... :-)

frnhr, 2021-01-06:
> Thanks, I reopened this issue.

@ivankravets The issue is still closed, though.

roosemberth, 2021-03-21:
I am also interested in this, for the same reasons as the original author of the issue.

KenwoodFox, 2023-02-07:
Im interested still but, because i would like to see a selfhosted option for hobbyists who host everything at home. is there no such self hosted broker?

AlexH0661, 2024-05-04:
Any updates on this?

sticilface, 2024-09-03:
I would also like this feature.