platformio/platformio-core #3693: Core 5.0.x can't handle platform-packages directive

orrmany, 2020-09-30:
### Configuration

**Operating system**: Windows 10 64bit

**PlatformIO Version** (`platformio --version`): 5.0.1.

### Description of problem
Platformio mishandles platform-packages directive

#### Steps to Reproduce

1. Create a new project with platformio.ini as follows:
`[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
monitor_speed = 115200
;platform_packages = framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble`
2. Add src/main.cpp with boiler plate blinky example:
`
#include <Arduino.h>
void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(100);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(100);
}` 
3. build your project. Observe that Platformio downloads and install the nRF55 platform with its toolchain, then it builds the project succesfully

4. Edit your platformio.ini. Remove the leading semicolon from the front of the line:
`platform_packages = framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble`  

5. attempt to build

### Actual Results
The build fails, the build script asserts (due to faulty download of wrong platform packages), like this:
```Error
AssertionError: :
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\main.py", line 194:
    target_elf = env.BuildProgram()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 661:
    return method(*args, **kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino.py", line 29:
    env.SConscript("arduino/adafruit.py")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino\adafruit.py", line 38:
    assert isdir(CORE_DIR)
```

### Expected Results
It should build normally using the nRF52 platform fork https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
`[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
monitor_speed = 115200
;platform_packages = framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(100);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(100);
}
```

### Additional info
Obviously, PlatformIO downloads some bogus "midi-test" v. 1.0.0. package instead of the nRF52 platform fork above

ivankravets, 2020-09-30:
It should be?
```
platform_packages = platformio/framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
```

orrmany, 2020-09-30:
I've changed to that, the results are similar:
```
> Executing task in folder nRF52custom: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
PLATFORM: Nordic nRF52 (4.4.1) > Adafruit Feather nRF52840 Express
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.0.0+sha.dda811e
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
AssertionError: :
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\main.py", line 194:
    target_elf = env.BuildProgram()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 661:
    return method(*args, **kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino.py", line 29:
    env.SConscript("arduino/adafruit.py")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino\adafruit.py", line 38:
    assert isdir(CORE_DIR)
==================================================================================== [FAILED] Took 0.82 seconds ====================================================================================
The terminal process "C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

orrmany, 2020-09-30:
then I've deleted the custom package dir manually, so it started to install: 

```
> Executing task in folder nRF52custom: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
git version 2.24.1.windows.2
Cloning into 'C:\Users\egbozie\.platformio\.cache\tmp\pkg-installing-lkdsalcj'...
remote: Enumerating objects: 909, done.
remote: Counting objects: 100% (909/909), done.
remote: Compressing objects: 100% (742/742), done.
remote: Total 909 (delta 256), reused 352 (delta 82), pack-reused 0 eceiving objects:  98% (891/909), 11.44 MiB | 5.67 MiB/s
Receiving objects: 100% (909/909), 16.25 MiB | 7.02 MiB/s, done.
Resolving deltas: 100% (256/256), done.
Updating files: 100% (726/726), done.
Submodule 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' (https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git) registered for path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'
Submodule 'libraries/Adafruit_nRFCrypto' (https://github.com/adafruit/Adafruit_nRFCrypto.git) registered for path 'libraries/Adafruit_nRFCrypto'
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-lkdsalcj/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'...
remote: Enumerating objects: 266, done.
remote: Counting objects: 100% (266/266), done.
remote: Compressing objects: 100% (156/156), done.
remote: Total 569 (delta 106), reused 183 (delta 76), pack-reused 303
Receiving objects: 100% (569/569), 464.96 KiB | 2.18 MiB/s, done.
Resolving deltas: 100% (243/243), done.
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-lkdsalcj/libraries/Adafruit_nRFCrypto'...
remote: Enumerating objects: 255, done.
remote: Counting objects: 100% (255/255), done.
remote: Compressing objects: 100% (114/114), done.
remote: Total 255 (delta 156), reused 227 (delta 133), pack-reused 0
Receiving objects: 100% (255/255), 345.65 KiB | 1.16 MiB/s, done.
Resolving deltas: 100% (156/156), done.
Submodule path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore': checked out '2f485087fd64d3fafffad414f0dc316c685d33d9'
Submodule path 'libraries/Adafruit_nRFCrypto': checked out '48b08a59d11b167c6b3c124db043a6df81cf5007'
[WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\cores\\nRF5\\linker' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\linker`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\cores\\nRF5' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\cores' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\cores`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\libraries' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\libraries`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52`
Error: Could not install package 'git+https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble' for 'windows_amd64' system
The terminal process "C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

orrmany, 2020-09-30:
The old core has installed the custom packages as ```.platformio/packages/framework-arduinoadafruitnrf52@src-86afce17ddb3b6c25ca6e2f29943ab03```. 
The new core did not created such a directory. Obviously it tries to install to  .platformio/packages/framework-arduinoadafruitnrf52, where the non-custom package resides, then it fails

ivankravets, 2020-10-16:
```
platform_packages = platformio/framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
```

Could you re-test with the latest `pio upgrade --dev`?

orrmany, 2020-10-16:
Will test soon, thank you. However, until then a comment: The new Core 5.x claims 100% backward compatibility with 4.x projects. Your request to add a "platformio/..." prefix breaks backward compatibility....

> 
> 
> It should be?
> 
> ```
> platform_packages = platformio/framework-arduinoadafruitnrf52 @ https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
> ```

orrmany, 2020-10-16:
Ok, it FAILs
here are the findings: 
- it either reports, or executes in the **wrong** project folder (I have a multiproject workspace)
- it gets assertion error

orrmany, 2020-10-16:
If I uncomment the `platform-packages` directive, then **it builds in the right directory** (`C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build\adafruit_feather_nrf52840`) , but **still reports wrong project dir** (it reports "SD-card-PoC", which project is also part of the same workspace. I do not know if it is relevantm, but "SD-card-PoC" happens to be located on Microsoft OneDrive and it is the "last" such OneDrive-located project in my workspace);

```
> Executing task in folder SD-card-PoC: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
PLATFORM: Nordic nRF52 (4.4.1) > Adafruit Feather nRF52840 Express
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.1900.200603 (19.0)
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52840\src\main.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\HardwarePWM.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\Print.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\Adafruit_USBD_CDC.cpp.o
Archiving .pio\build\adafruit_feather_nrf52840\libFrameworkArduinoVariant.a
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\Adafruit_USBD_Device.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\cdc\cdc_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\hid\hid_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\midi\midi_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\msc\msc_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\usbtmc\usbtmc_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\class\vendor\vendor_device.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\common\tusb_fifo.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\device\usbd.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\device\usbd_control.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\microchip\samd\dcd_samd.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\hal_nrf5x.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\tusb.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\TinyUSB\Adafruit_TinyUSB_nRF.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\Uart.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\WInterrupts.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\WString.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\abi.cpp.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\delay.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\freertos\Source\croutine.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\freertos\Source\event_groups.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\freertos\Source\list.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\freertos\Source\portable\MemMang\heap_3.c.o
Compiling .pio\build\adafruit_feather_nrf52840\FrameworkArduino\freertos\Source\queue.c.o
[... cut ...]
RAM:   [          ]   2.7% (used 6612 bytes from 248832 bytes)
Flash: [          ]   4.5% (used 36540 bytes from 815104 bytes)
Building .pio\build\adafruit_feather_nrf52840\firmware.zip
Zip created at .pio\build\adafruit_feather_nrf52840\firmware.zip
=============================================================================== [SUCCESS] Took 30.57 seconds ===============================================================================

Terminal will be reused by tasks, press any key to close it.
```

orrmany, 2020-10-16:
Note, that PlatformIO _did not_ attempt to installt the correct platform-package from github, maybe the stale bogus package, which contains some MIDI-test shit is cause of the problem?

orrmany, 2020-10-16:
Ok, after manually removing the stale custom-package via PIO-Home UI, the bogus MIDI-test package has gone.  Platformio now started to install the custom-packages, **but again messed up with the non-custom variant of the same library**. It seemingly tried to install ON TOP of the existing (non-custom) arduinoadafruitnrf52 framework. **I as a PlatformIO user want/need to use _both_  framework packages _simultaneously_**.  Installing one on top of  other is not acceptable.

```
> Executing task in folder SD-card-PoC: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
git version 2.24.1.windows.2
Cloning into 'C:\Users\egbozie\.platformio\.cache\tmp\pkg-installing-4ydr6buj'...
remote: Enumerating objects: 909, done.
remote: Counting objects: 100% (909/909), done.
remote: Compressing objects: 100% (742/742), done.
remote: Total 909 (delta 256), reused 352 (delta 82), pack-reused 0R
Receiving objects: 100% (909/909), 16.25 MiB | 14.42 MiB/s, done.
Resolving deltas: 100% (256/256), done.
Updating files: 100% (726/726), done.
Submodule 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' (https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git) registered for path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'
Submodule 'libraries/Adafruit_nRFCrypto' (https://github.com/adafruit/Adafruit_nRFCrypto.git) registered for path 'libraries/Adafruit_nRFCrypto'
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-4ydr6buj/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'...
remote: Enumerating objects: 266, done.
remote: Counting objects: 100% (266/266), done.
remote: Compressing objects: 100% (156/156), done.
remote: Total 569 (delta 106), reused 183 (delta 76), pack-reused 303
Receiving objects: 100% (569/569), 464.96 KiB | 3.16 MiB/s, done.
Resolving deltas: 100% (243/243), done.
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-4ydr6buj/libraries/Adafruit_nRFCrypto'...
remote: Enumerating objects: 255, done.
remote: Counting objects: 100% (255/255), done.
remote: Compressing objects: 100% (114/114), done.
remote: Total 255 (delta 156), reused 227 (delta 133), pack-reused 0
Receiving objects: 100% (255/255), 345.65 KiB | 3.36 MiB/s, done.
Resolving deltas: 100% (156/156), done.
Submodule path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore': checked out '2f485087fd64d3fafffad414f0dc316c685d33d9'
Submodule path 'libraries/Adafruit_nRFCrypto': checked out '48b08a59d11b167c6b3c124db043a6df81cf5007'
[WinError 5] Access is denied: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\libraries\\Bluefruit52Lib' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib`
[WinError 5] Access is denied: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\libraries\\Bluefruit52Lib' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Bluefruit52Lib`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52\\libraries' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52\libraries`
[WinError 145] The directory is not empty: 'C:\\Users\\egbozie\\.platformio\\packages\\framework-arduinoadafruitnrf52' 
Please manually remove the file `C:\Users\egbozie\.platformio\packages\framework-arduinoadafruitnrf52`
Error: Traceback (most recent call last):
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 143, in cli
    is_test_running,
  File "c:\users\egbozie\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
[... cut ...]
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

orrmany, 2020-10-16:
ok, I have restarted the PlatformIO-IDE, which cleared the file-access problem. Obviously that was some _multihread race condition_, i.e., you need to check your semaphores...
After the restart the custom-framework _installation_ did not report error anymore, but **it installed the same MIDI-test crap**, so then the build has failed:
```
> Executing task in folder SD-card-PoC: C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe run <

Processing adafruit_feather_nrf52840 (platform: nordicnrf52; board: adafruit_feather_nrf52840; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble
git version 2.24.1.windows.2
Cloning into 'C:\Users\egbozie\.platformio\.cache\tmp\pkg-installing-12ftba9g'...
remote: Enumerating objects: 909, done.
remote: Counting objects: 100% (909/909), done.
remote: Compressing objects: 100% (742/742), done.
Receiving objects:  98% (891/909), 15.07 MiB | 15.05 MiB/s
Receiving objects: 100% (909/909), 16.25 MiB | 15.32 MiB/s, done.
Resolving deltas: 100% (256/256), done.
Updating files: 100% (726/726), done.
Submodule 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore' (https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore.git) registered for path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'
Submodule 'libraries/Adafruit_nRFCrypto' (https://github.com/adafruit/Adafruit_nRFCrypto.git) registered for path 'libraries/Adafruit_nRFCrypto'
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-12ftba9g/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore'...
remote: Enumerating objects: 266, done.
remote: Counting objects: 100% (266/266), done.
remote: Compressing objects: 100% (156/156), done.
remote: Total 569 (delta 106), reused 183 (delta 76), pack-reused 303
Receiving objects: 100% (569/569), 464.96 KiB | 4.39 MiB/s, done.
Resolving deltas: 100% (243/243), done.
Cloning into 'C:/Users/egbozie/.platformio/.cache/tmp/pkg-installing-12ftba9g/libraries/Adafruit_nRFCrypto'...
remote: Enumerating objects: 255, done.
remote: Counting objects: 100% (255/255), done.
remote: Compressing objects: 100% (114/114), done.
remote: Total 255 (delta 156), reused 227 (delta 133), pack-reused 0
Receiving objects: 100% (255/255), 345.65 KiB | 380.00 KiB/s, done.
Resolving deltas: 100% (156/156), done.
Submodule path 'cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore': checked out '2f485087fd64d3fafffad414f0dc316c685d33d9'
Submodule path 'libraries/Adafruit_nRFCrypto': checked out '48b08a59d11b167c6b3c124db043a6df81cf5007'
Tool Manager: framework-arduinoadafruitnrf52 @ 1.0.0+sha.dda811e has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52840.html
PLATFORM: Nordic nRF52 (4.4.1) > Adafruit Feather nRF52840 Express
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 1.0.0+sha.dda811e
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
AssertionError: :
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\main.py", line 194:
    target_elf = env.BuildProgram()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egbozie\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 661:
    return method(*args, **kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egbozie\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino.py", line 29:
    env.SConscript("arduino/adafruit.py")
[... cut ...]
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egbozie\.platformio\platforms\nordicnrf52\builder\frameworks\arduino\adafruit.py", line 38:
    assert isdir(CORE_DIR)
================================================================ [FAILED] Took 54.69 seconds ================================================================
The terminal process "C:\Users\egbozie\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

orrmany, 2020-10-16:
This crap has been installed instead of the real packages:
[framework-arduinoadafruitnrf52.zip](https://github.com/platformio/platformio-core/files/5391783/framework-arduinoadafruitnrf52.zip)

orrmany, 2020-10-16:
I looked around a little bit:

- `.platformio/platforms/` content looks good:
  - `nordicnrf52`, i.e., the original platform 
  - `nordicnrf52@src-5bf0db87290a59bacd6fcea330624e29`, i.e., the customized platform
- `.platformio/packages/` content does **_not_** look good:
  - although `framework-arduinoadafruitnrf52` is present, but contains the MIDI-test crap attached in the earlier comment. _It should have continued to contain_ the original packages
  - `framework-arduinoadafruitnrf52@src-5bf0d...` is **missing**, should have contain the freshly installed custom packages from github

ivankravets, 2020-10-29:
> https://github.com/orrmany/Adafruit_nRF52_Arduino.git#develop-ant-plus-ble

I don't see `package.json` in this branch.

closed 2020-10-29 by ivankravets

orrmany, 2020-10-29:
@ivankravets, it might be so, I will check upstream. However, does that justify to install some MIDI-test crap instead???

ivankravets, 2020-10-29:
Yes, that is a reason. The package manager looks recursively for `package.json` manifest which says that this is a root of a package. Please check any official package for the format of `package.json`.

orrmany, 2020-10-29:
By all respect, I disagree @ivankravets. That might _trigger_ the bug in the pkg. management, but it is till a bug, nonetheless. Silenltly _installing_ some nonsense due to a broken package content and then _later_ failing with misleading compile time messages --- that is a bug, IMHO.

ivankravets, 2020-10-29:
Manifest is the only genuine information. People keep packages in the different strucutres. We recommend to keep manifest in the root. Others prefer to have an intermediary folder between root.

So, if you have package.json in the root and it does not work - please open issue. We are not responsible for the packages without manifests or with broken.

orrmany, 2020-10-29:
ok, so then the trigger for the bug is that I've referenced a clone of the _original _ Adafruit framework, not the clone of the PlatformIO package...  This worked fine with 4.xPlatformIO :/
I will see what I can do for workaround...

cujomalainey, 2020-10-29:
I'm sorry I'm a little lost here, do we have a work around for why forking a BSP does not work?

ivankravets, 2020-10-29:
It works. Just fork it and add `package.json` into the root.

ivankravets, 2020-10-29:
See example https://github.com/espressif/arduino-esp32/blob/master/package.json

orrmany, 2020-10-30:
@ivankravets: I tried to check the package.json for the non-modified Adafruit_nRF52_Arduino platform package. IIRC the command for this in 5.0.x: `pio access list` according to https://docs.platformio.org/en/latest/core/userguide/access/cmd_list.html
However: contrary to documentation such command **does not exist** in 5.0.b2: 
```
PS C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build> pio  access --help
Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "access"
PS C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build> pio  --version
PlatformIO, version 5.0.2b2
PS C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build>
```

ivankravets, 2020-10-30:
Wow! How it's possible? Could you try `pio upgrade --dev`?

orrmany, 2020-10-30:
@ivankravets, nope: 
```
PS C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build> pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.0.2rc1
Release notes: https://docs.platformio.org/en/latest/history.html
Warning! Please restart IDE to affect PIO Home changes
PS C:\Users\egbozie\local\CustomBlinky\CustomBlinky\.pio\build> pio  access --help
Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "access"
```

orrmany, 2020-10-30:
Restarting the IDE did not help either:
```
PS C:\Users\egbozie\Google Drive\Arduino\Programming\GitHub\ANTplus-PoC> pio  access --help
Usage: pio [OPTIONS] COMMAND [ARGS]...
Try 'pio -h' for help.

Error: No such command "access"
PS C:\Users\egbozie\Google Drive\Arduino\Programming\GitHub\ANTplus-PoC> pio  --version
PlatformIO, version 5.0.2rc1
PS C:\Users\egbozie\Google Drive\Arduino\Programming\GitHub\ANTplus-PoC>
```

ivankravets, 2020-10-30:
Sorry, no idea. Thoundsands of develoeprs use PlatformIO everyday. You are the first with this issue. Let's wait for more reports. I don't know what I can advice. Windows brings a ton of problems latest times. Alsom people install spy "antivirus" software which blocks Python.

Try to remove `%HOMEDIR%/penv` folder and restart VSCode. We will not provide more help here. Please use https://community.platformio.org/

We discuss here PlatformIO Core issues.

cujomalainey, 2020-10-30:
@orrmany i can verify on my end it works on OSX

ivankravets, 2020-10-30:
Great! How did you fix it?

cujomalainey, 2020-10-30:
I haven't done full test yet, just verified the access command is there (didn't do anything) will hopefully try fixing build tommorrow

dmpolukhin, 2020-11-09:
@ivankravets It looks like it is some deadlock here. Adafruite doesn't want to add package.json, see https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/571 and I see their point about not updating version there. PlatformIO insists on the file that was not required before. So what is the solution - only fork and add the file?

ivankravets, 2020-11-09:
Yes, just fork and put `package.json` to the root of the repository.

orrmany, 2020-11-10:
@ivankravets, in other words: you require that the package metadata be part of the package. That is a problem for all of your package maintainers with 3rd pty libraries. Now they have create and maintain a fork just for the sake of adding a metadata...

ivankravets, 2020-11-10:
Yes, that was the step to remove these issues when people try to install "hello-monkey" as a dev-platform. Then contact us and ask why the monkey does not speak. So, let's protect beginners from monkeys and improve our UX.

orrmany, 2020-11-10:
@ivankravets , a looser question maybe: how can I find the _URL_ of the repo containing the platformio framework packages, i.e., where PlatformIO team maintains _the PlatformIO fork_  of the Adafruit framework library?  The repo, where the PlatformIO team has added your version of package.json to the Adafruit code? 

I would like to study your fork (and your other package-forks, as well)
Thanks!

ivankravets, 2020-11-10:
We don't have forks because we don't modify official software. The only what we do - we put `package.json` and fill manifest with metadata. [Some packages](https://bintray.com/beta/#/platformio/tool-packages?tab=packages) are mirrored on Bintray. We plan to release our new front-end to the new registry soon.

orrmany, 2020-11-10:
Is there any step-by-step tutorial,  how can an innocent user, like me, who has had a valid solution for his custom framework hack in PlatformIO 4.x which _just had worked_, migrate to 5.x?  

The only reason I have had to fork the Adafruit framework is for adding ANT+ radio support, which Adafruit cannot embrace due to legal reasons. The guy who has originally made the PlatformioIO integration hack for me has gone long ago. 

I am extremely frustrated: I have had a setup that _worked_  in 4.x and _5.x  just broke it_and I am lost now. I do not want to learn how to develop/hack PlatformIO, I just want to _use it_

I have read https://docs.platformio.org/en/latest/core/migration.html but it is _no help_. That document sucks for non-insiders, I am sorry: it describes _what has been changed_ but gives zero clue for me _how_ shall I fix my previously working stuff... 

I do not want to spend days on learning PlatformIO internals, I couldn't care less. It is an extremely frustrating situation and it directly invalidates the original promise of PlatformIO: make an dev.env.,  which "just works"... :(

ivankravets, 2020-11-10:
@orrmany Very good question! So, found the root why do you need to work official package. Could check these docs https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files ?

If it is not clear, we would be happy to improve.

So, the goal is that you will not worry about the framework package. You will just patch it to your needs. This will help you to be up-to-date with the latest framework and apply your changes.

orrmany, 2020-11-10:
Ok, thanks! Considering that adding the ANT+ radio support is not a simple diff, but in fact a bunch of additional files and several modifications to the original files, so I use git and a forked repo. 
I hope this "patching" works with somehow with git. 

Do you suggest that 
1.  I shall compare the upstream library repo to my fork (which I regularly sync with the upstream) using git and dump the multi-file difference into a "patch-tarball"
2.  add the "patch-tarball" to my PlatformIO project, which refers to the original, non-forked library from Adafruit as a dependency
3.  and somehow configure PlatformIO to redo my fork during build time by patching the upstream library with the patch generated in step 1 from my fork elsewhere?

orrmany, 2020-11-10:
> So, the goal is that you will not worry about the framework package. You will just patch it to your needs. This will help you to be up-to-date with the latest framework and apply your changes.

I have to say I need to contest what you write here, @ivankravets. What helps me to be up-to-date with the latest framework and apply my changes is to keep my fork in sync with upstream. I have to do that regardless of PlatformIO both for my sake and for the sake of my GitHub downstream followers.

Requiring me to _redo_ this syncig in the PlatformIO build system really sounds as pure waste. I fail to understand why is it "easier" than I just being able to tell PlatformIO to "use this alternative forked repo instead of the upstream repo"?? 

What changes I've made to my fork w.r.t to the upstream library shall _not_ be the business of the PlatformIO build system in any way, IMHO...

ivankravets, 2020-11-10:
> than I just being able to tell PlatformIO to "use this alternative forked repo instead of the upstream repo"??

You can, I still do not understand your issue. Everything works as expected. We just fixed a historical issue. If the PlatformIO package manager does not want to install an unknown package it does not mean that we broke something.

[Over  1,000 issues](https://github.com/platformio/platformio-home/issues) related to "this freedom" which we had in PlatformIo Core 4.0.

cujomalainey, 2020-11-10:
So you are saying this was never feature but an unfixed bug that we could just pass in a platform-packages repo that was simply a fork as an override?

ivankravets, 2020-11-10:
> we could just pass in a platform-packages repo that was simply a fork as an override?

You can do this with PlatformIO 5 as well. It has full support to override the default package with a package from a git.

orrmany, 2020-11-18:
Finally made the build working again under Core 5.0.3 by adding a package.json _and_ creating a platform.json, as well, for my platform fork, too.  (I had to create a platform fork as I wanted to create a new board type for the modified firmware.
I had to use CLI, though, as the PlatformIO IDE was very unstable with the (faulty) repos: I got constantly hanging build sub-processes during the (failing) package installation, without any meaningful printout.
The package management at the time of writing this (Core v. 5.0.3) is _absolutely_ not robust, it can't handle packaging errors gracefully, nor to give any meaningful error message. 
Now the issue is closed for me, as well. :)