platformio/platformio-core #3700: Python 3.9 is not supported

dariomas, 2020-10-09:
### Configuration

**Operating system**: Linux Fedora 33

**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.1

### Description of problem

The Platformio installer on a Linux system wit Python 3.9 (Fedora 33) hung while installing because it does not find the
'platformio/contrib-pysite @ ~2.39.0'

### Actual Results

VSCode: 1.50.0
PIO IDE: v2.1.0
System: Linux, 5.8.13-300.fc33.x86_64, x64

%23 Exception

Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'linux_x86_64'

### Additional info

A temporary resolution:

- setting in VSCode path https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-custompath
`"platformio-ide.customPATH": "/home/xxxx/.platformio/penv/bin/"`

- disabling in VSCode builtin core https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpiocore

- installing python3.8 in Fedora 33
`dnf install python3.8`

run the Platformio installation 

- script with python 3.8 (creating virtualenv with python3.8)
`python3.8 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"`

- check into the `penv/bin/` that python links to 3.8

%23 Configuration

VSCode: 1.50.0
PIO IDE: v2.1.0
System: Linux, 5.8.13-300.fc33.x86_64, x64

bronsen, 2020-10-19:
Another approach is to create a python virtualenv with the desired python version. Then start vscode from the actived virtualenv: platformio then picks ups the compatible version.

[Here are some more details](https://infuanfu.de/howto/2020/10/19/platformio-vscode-python.html), for fish shell, using pyenv and virtualfish.

ivankravets, 2020-10-20:
Packages for Python 3.9 are published in the registry.

commit 23e95965 (2020-10-20): Automatically build PlatformIO Core extra Python dependencies on a host machine if they are missed in the registry // Resolve #3700
closed 2020-10-20 by ivankravets