platformio/platformio-core #3702: Menuconfig ignores library Kconfig files

tuupola, 2020-10-15:
What kind of issue is this?

- [*] **Feature Request**.

------------------------------------------------------------------

### Configuration

**Operating system**: macOS Mojave 10.14.6 
**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.1

### Description of problem

Menuconfig ignores library KConfig files unless the library is installed in `components` folder. Tested with `esp-idf`.

#### Steps to Reproduce

1. `$ pio run -t menuconfig`

### Actual Results

Library config options are not shown.

### Expected Results

Library config options should be  shown.

### Additional info

Workaround is to pass the `--storage-dir` option when installing the library.

```
$ pio lib  --storage-dir components install foo/bar
```

However this fails for future users running `pio run` because dependencies are downloaded into `.pio/environment` folder. Another workaround is to use a `platformio.ini` which abuses the environment name:

```
[platformio]
libdeps_dir = .

[env:components]
platform = espressif32
board = m5stack-grey
framework = espidf
lib_deps = foo/bar
```

Downside for this approach is that you can only only one environment and it has to be called `components`.

ivankravets, 2020-10-20:
You can override a name of `libdeps` dir https://docs.platformio.org/en/latest/projectconf/section_platformio.html#libdeps-dir

So, did you try `libdeps_dir = components`?

ivankravets, 2020-10-28:
Please reopen if you still need help

closed 2020-10-28 by ivankravets

nicklasb, 2024-02-25:
To update this one a little bit:
I had to add "./" to run from the project library, and you need to add the folder first:
`pio pkg install --storage-dir ./components  --library "robusto/robusto"`