platformio/platformio-core #3709: Python was not found but can be installed from the Micros

a97077088, 2020-10-22:
What kind of issue is this?

- [ YES] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ YES] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ YES] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64，python3.9(x64)

**PlatformIO Version** (`platformio --version`):PlatformIO, version 5.0.2b2

### Description of problem

#### Steps to Reproduce

1.platformio.exe -c clion run --target upload -e stc89c52rc
2.
3.

### Actual Results

Processing stc89c52rc (platform: intel_mcs51; board: stc89c52rc)
------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/intel_mcs51/stc89c52rc.html
PLATFORM: Intel MCS-51 (8051) (1.2.2) > Generic STC89C52RC
HARDWARE: STC89C52RC 11MHz, 512B RAM, 8KB Flash
PACKAGES:
 - tool-stcgal 1.104.0 (1.4)
 - toolchain-sdcc 1.30804.10766 (3.8.4)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\stc89c52rc\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Flash: [          ]   1.3% (used 103 bytes from 8192 bytes)
Configuring upload protocol...
AVAILABLE: stcgal
CURRENT: upload_protocol = stcgal
Looking for upload port...
Auto-detected: COM36
Uploading .pio\build\stc89c52rc\firmware.hex
Python was not found but can be installed from the Microsoft Store: https://go.microsoft.com/fwlink?linkID=2082640scons:
 *** [upload] Error 9009
============================================== [FAILED] Took 0.82 seconds ==============================================

### Expected Results

He can't recognize python well, my machine is installed python3.9 and has done env

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:stc89c52rc]
platform = intel_mcs51
board = stc89c52rc
```

**Source file to reproduce issue:**
```cpp
#include <8051.h>
int main() {
// write your code here
}

```

### Additional info

chenwangnec, 2020-10-28:
Wèntí jiějué fāngfǎ

xiūgǎi C:\Users\yònghù míng\.Platformio\platforms\intel_mcs51\builder
main.Py

156 xíng

UPLOADCMD="python3%s $UPLOADERFLAGS $SOURCE" % stcgal)

xiūgǎi wèi

UPLOADCMD="python%s $UPLOADERFLAGS $SOURCE" % stcgal)
biānyì shàngchuán shāo kǎo 
wǒ de jiùshì zhèyàng jiějué de
展开
217/5000
Problem solving method

Modify C:\Users\user name\.platformio\platforms\intel_mcs51\builder
main.py

Line 156

UPLOADCMD="python3 %s $UPLOADERFLAGS $SOURCE"% stcgal)

change into

UPLOADCMD="python %s $UPLOADERFLAGS $SOURCE"% stcgal)
Compile upload burn copy
Mine is solved like this

chenwangnec, 2020-10-28:
Problem solving method

Modify C:\Users\user name\.platformio\platforms\intel_mcs51\builder
main.py

Line 156

UPLOADCMD="python3 %s $UPLOADERFLAGS $SOURCE"% stcgal)

change into

UPLOADCMD="python %s $UPLOADERFLAGS $SOURCE"% stcgal)
Compile upload burn copy
Mine is solved like this

翻译中文

问题解决方法

修改C:\Users\用户名\.platformio\platforms\intel_mcs51\builder
main.py

156行

UPLOADCMD="python3 %s $UPLOADERFLAGS $SOURCE" % stcgal )

修改为

UPLOADCMD="python %s $UPLOADERFLAGS $SOURCE" % stcgal )
编译上传烧拷 
我的就是这样解决的 我的环境也是winows 10 vscode+PlatformIO 8051-stc89c52rc
已经正常编译和烧拷固件